ct-Open-Source/tuya-convert #499: can not connect with mobile to vtrust-flash ssid

riker65, 2020-01-12:
Hi

running the proces on a respberry.

vtrust-flash hotspot ap is coming up

but can not connect to it with the mobile

raspberry crashes, can not enter anything on the keyboard, seems that hotspot is still up but no connection possible

what am I missing?

thanks for your support

kueblc, 2020-01-12:
Could be #409 or #373

riker65, 2020-01-13:
ok thanks will check

where do I find the log files?

I was working on the raspberry without ssh.

ap vtrusth-flush is up but cvonfig is still 

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.139  netmask 255.255.240.0  broadcast 192.168.15.255
        inet6 fe80::76da:38ff:fe7f:b271  prefixlen 64  scopeid 0x20<link>
        ether 74:da:38:7f:b2:71  txqueuelen 1000  (Ethernet)
        RX packets 2834  bytes 378263 (369.3 KiB)
        RX errors 0  dropped 7  overruns 0  frame 0
        TX packets 36  bytes 5954 (5.8 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

Thanks

riker65, 2020-01-14:
Hi still can no connect

after deleting of the modprob exclude and wlan setting at least I get the ip config required

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.42.42.1  netmask 255.255.255.0  broadcast 0.0.0.0
        inet6 fe80::76da:38ff:fe7f:b271  prefixlen 64  scopeid 0x20<link>
        ether 74:da:38:7f:b2:71  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 10159  bytes 1846009 (1.7 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

But mobile can not connect

where a the logfiles?

kueblc, 2020-01-14:
After you pulled the latest from git, did you [uninstall sslpsk and run install_prereq again](https://github.com/ct-Open-Source/tuya-convert/releases/tag/v2.4.1)?

riker65, 2020-01-14:
Hi, how to  uninstall sslpsk?

I will do a clean install again

riker65, 2020-01-14:
> After you pulled the latest from git, did you [uninstall sslpsk and run install_prereq again](https://github.com/ct-Open-Source/tuya-convert/releases/tag/v2.4.1)?

no just a new pull, so will do a new clean install

kueblc, 2020-01-14:
`sudo -H pip3 uninstall sslpsk`
`sudo -H python3 -m pip uninstall sslpsk`

riker65, 2020-01-14:
ok thanks will try

riker65, 2020-01-14:
Hi Colin,

after running install_prereq again. I got :
```

Installing collected packages: sslpsk
  Running setup.py install for sslpsk ... done
Successfully installed sslpsk-1.0.0
Ready to start upgrade
pi@RPI2_ST_PX_157:~/tuya-convert $ sudo apt-get upgrade
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
Paketaktualisierung (Upgrade) wird berechnet... Fertig
Die folgenden Pakete sind zurückgehalten worden:
  wolfram-engine
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 1 nicht aktualisiert.
```
running apt upgrade, -> noting to upgrade? 

so what is expected to do after this message?

```
Successfully installed sslpsk-1.0.0
Ready to start upgrade
```

riker65, 2020-01-14:
so now , I think I have a clean new installatoion.

But mobile is not connecting to ssid vtrust-flash

see logs

Thanks for the great support!!

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4060761/smarthack-mqtt.log)

[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4060762/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4060763/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4060764/smarthack-wifi.log)

kueblc, 2020-01-14:
No psk log?

riker65, 2020-01-15:
Hi
after redoing it all, new install, it worked.

thanks a lot

but after connecting to espurna, connecting to it and setting new password and wifi.
I can not connect to espurna - hotsport not coming up and 
espurna switch not connecting to local wlan.

just any idea?

thanks a lot

riker65, 2020-01-15:
actually tried it again in the morning
connection mobile to vtrsut-flash is again not working.

please find all logs here.

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4064007/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4064008/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4064009/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4064010/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4064011/smarthack-wifi.log)

kueblc, 2020-01-15:
Have you checked for a firewall? #450 #464

danielengels, 2020-01-15:
Same problem here with a clean installation on a raspberry pi 3 with rasbian (buster lite). I try several devices but no vtrust network appears on them.
I was thinking may be something with the wifi but as a client, the raspberry can connect to a network.
After a reinstall there is still no open AP to connect from mobile. Firewall seems not to work too as i try to stop both ways via initd and systemctl
`
Checking for network interface wlan0... Found.
Checking UDP port 53... Occupied by dnsmasq with PID 510.
Port 53 is needed to resolve DNS queries
Do you wish to terminate dnsmasq? [y/N] y
Attempting to stop dnsmasq.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Occupied by mosquitto with PID 499.
Port 1883 is needed to run MQTT
Do you wish to terminate mosquitto? [y/N] y
Attempting to stop mosquitto.service
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen...........
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue`

before and after pressing ENTER no AP appears on serveral devices. After some retries, here are the logs.

[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4066443/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4066444/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4066445/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4066446/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4066447/smarthack-mqtt.log)

danielengels, 2020-01-15:
my problem is solved - sorry. I tried the same card on a same raspberry pi model and it works first time... so my other raspberry pi seems to have a problem with the wifi chip :S

riker65, 2020-01-16:
It did work after some reinstallations

thanks perfect

closed 2020-01-16 by riker65