ct-Open-Source/tuya-convert #500: PSK str/bytes issue

gmckeown, 2020-01-13:
**Product:** Teckin SL02 Smart Light Strip (non-idividually-addressable LED strip).
**Issue:**

Most obvious error is in the PSK log:

```
new client on port 443 from 10.42.42.20:53277
Traceback (most recent call last):
  File "./psk-frontend.py", line 111, in <module>
    main()
  File "./psk-frontend.py", line 107, in main
    p.data_ready_cb(s)
  File "./psk-frontend.py", line 78, in data_ready_cb
    self.new_client(_s)
  File "./psk-frontend.py", line 66, in new_client
    hint=self.hint.decode())
  File "/usr/local/lib/python3.7/dist-packages/sslpsk/sslpsk.py", line 104, in wrap_socket
    _ssl_set_psk_server_callback(sock, cb, hint)
  File "/usr/local/lib/python3.7/dist-packages/sslpsk/sslpsk.py", line 80, in _ssl_set_psk_server_callback
    _      = _sslpsk.sslpsk_use_psk_identity_hint(_sslobj(sock), hint if hint else b"")
TypeError: a bytes-like object is required, not 'str'
```

**Platform:** Running scripts on Raspberry Pi 3B with Ubuntu 19.04 64-bit.

Device contains board ref [TYWE3S](https://fccid.io/2AFNL-TYWE3S/User-Manual/Users-Manual-3525098), which does appear to be ESP2866-based.

[image]

Script logs:

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4051762/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4051763/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4051764/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4051765/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4051766/smarthack-wifi.log)

Seems to be related to #496 -- if I revert this change, I'm able to flash the device.

rhyvin, 2020-01-13:
I also reverted the previous change, I am running Raspbian 10 (buster), I was attempting to flash a Merkury A21 Colored bulb. After I reverted the change made in #496 I was able to flash the device.

jjtParadox, 2020-01-13:
Error from #497 says the arg needs to be a string, this error says the arg needs to be bytes... What is it supposed to be????

jjtParadox, 2020-01-13:
Ah, looks like my PR was based on an older version of sslpsk that didn't use byte hints. I could have sworn I ran the install prereqs script when I checked out the dev branch...

kueblc, 2020-01-13:
Thanks @gmckeown and @jjtParadox for sorting this out!

commit b9d4cdc6 (2020-01-13): Revert #496 fixes #500
commit 6febeec3 (2020-01-13): Merge pull request #501 from ct-Open-Source/development

Revert #496 fixes #500
closed 2020-01-13 by kueblc