ct-Open-Source/tuya-convert #502: Gosund SP112 - Device did not appear with the intermediate firmware

EnidanBri, 2020-01-13:
Hi all, 

I am trying to flash a Gosund SP112 switch OTA. 

I am getting so far that I started the ./start_flash.sh on the raspi via SHH (connected from my laptop via eth0), have my smartphone connected to the WIFI network vtrust-flash when the script asks me to do so. I put my SP112 device into pairing mode (fast blinking), and when I hit Enter in the script, the LED of the SP112 changes from blinking fast blue to constant red. So 'something' is happening, but apparently, it doesn't receive the new firmware. 
After an unsuccessful attempt, I can still connect it to the Gosund app, so nothing has changed on the device. I had it paired via the Gosund app earlier, but I removed it in the app before trying to flash.

The device ID is 82008006600194c2f3f3
MAC address: 60:01:94:c2:f3:f3
Firmware according to Gosund app is the most recent: 1.0.0 

I already looked at a number of issues here, but none of them was exactly the same, I think. 

Any ideas?

Thanks & Regards,
Nadine

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4056106/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4056107/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4056108/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4056109/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4056110/smarthack-wifi.log)

kueblc, 2020-01-13:
Could be a firewall issue #450 #464

riker65, 2020-01-14:
edit same situation for me.

Tuya light turned from binking fast into led static 

but error occured.

`wlan0: interface state ENABLED->DISABLED
wlan0: AP-DISABLED 
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
Restarting NetworkManager...
Failed to restart network-manager.service: Unit network-manager.service not found.
System info
===========
6febeec
Linux RPI2_ST_PX_157 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
OpenSSL 1.1.0l  10 Sep 2019
Dnsmasq Version 2.76  Copyright (c) 2000-2016 Simon Kelley
Kompilierungs-Optionen IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify

Für diese Software wird ABSOLUT KEINE GARANTIE gewährt.
Dnsmasq ist freie Software, und du bist willkommen es weiter zu verteilen
unter den Bedingungen der GNU General Public Lizenz, Version 2 oder 3.
hostapd v2.6
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2016, Jouni Malinen <j@w1.fi> and contributors
Python 3.5.3
===========
Stopping NetworkManager...
Failed to stop network-manager.service: Unit network-manager.service not loaded.
Configuring AP interface...
RTNETLINK answers: File exists
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
Using interface wlan0 with hwaddr 74:da:38:7f:b2:71 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: AP-STA-CONNECTED d8:6c:02:fc:fa:83
wlan0: AP-STA-CONNECTED d8:6c:02:fc:fa:83
wlan0: AP-STA-CONNECTED d8:f1:5b:b7:41:d8
wlan0: AP-STA-CONNECTED d8:6c:02:fc:fa:83
wlan0: AP-STA-CONNECTED d8:6c:02:fc:fa:83
wlan0: AP-STA-CONNECTED d8:6c:02:fc:fa:83
wlan0: AP-STA-CONNECTED d8:6c:02:fc:fa:83
wlan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED d8:f1:5b:b7:41:d8
wlan0: AP-STA-DISCONNECTED d8:6c:02:fc:fa:83
wlan0: AP-DISABLED 
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
Restarting NetworkManager...
Failed to restart network-manager.service: Unit network-manager.service not found.`

EnidanBri, 2020-01-14:
Hello again,

no, it was not the firewall. When I tried to stop the firewall service, as suggested in the other threads, I received the message that there is no firewall service.

However, it was true that I had trouble a) finding and b) connecting to the vftrust-flash wifi (like it was in the linked issue).
When I first tried to connect to the wifi with a fresh Rasbian buster lite install on a Raspberry Pi 3B+, I received a message that the AP mode is not supported. 
And then I tried all kinds of things, including using a separate wifi dongle, renaming the wlans on the Raspberry, renaming them back, unplugging the external wifi dongle again, setting up the internal one to be able to do AP mode (which I did not know was possible earlier), and then assigning static IP addresses to the internal wifi dongle to be able to obtain an IP address, which in the end resulted in me being able to connect to the vftrust-flash wifi with my phone and receiving an IP address.

However, this morning I noticed that my phone got the IP address 10.42.42.12 after connecting to the wifi, and my laptop got an IP 192.168.1.140, which was part of the IP area which I used when trying to assign static IPs to be able to connect. This made me think that I somehow screwed up something, and I restarted from scratch and flashed the raspi with a clean Raspbian buster lite again.

**And miraculously, it worked without any issues with that clean install!** I also did not get the "AP mode not supported" message which I received after the first clean install.
So, I can't really tell what I did do differently this time. I also used a fresh install of Raspbian on my first attempt, and followed the instructions (https://www.tasmota.info/ota/) step by step. But obviously, something **was** different.

I am only describing this in case anybody encounters the same issues...

Thanks anyway for the help,
Kind Regards,
Nadine

closed 2020-01-14 by EnidanBri