ct-Open-Source/tuya-convert #503: BlitzWolf® BW-SHP6 issue

jokerigno, 2020-01-15:
Hi,

I have issue flashing 2 shp6 plugs.

I tried with version 1 first and now with version 2. Same failure.

One device now stil can be put on fast blinking mode, the other one has no led reaction.

This is what I see in console:

`Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101

IoT-device is online with ip 10.42.42.42
Stopping smart config
Fetching firmware backup
./start_flash.sh: line 106:  4240 Terminated              ./smartconfig/main.py
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  1 1024k    1 11680    0     0     21      0 13:52:12  0:08:55 13:43:17     0Checking for network interface wlan0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fa                                                                                                                                                             st). This is usually done by pressing and holding the primary button of the devi                                                                                                                                                             ce
   Make sure nothing else is plugged into your IoT device while attempting to fl                                                                                                                                                             ash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101

IoT-device is online with ip 10.42.42.42
Stopping smart config
Fetching firmware backup
./start_flash.sh: line 106:  4240 Terminated              ./smartconfig/main.py
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
`

these are logs:

web log:

[... 586 lines cut ...]
1579100739: Opening ipv4 listen socket on port 1883.
1579100739: Opening ipv6 listen socket on port 1883.
1579100760: New connection from 10.42.42.1 on port 1883.
1579100760: New client connected from 10.42.42.1 as 10343760ecfabc343234 (c1, k60, u'WCofXic7RAXrTpYtYDA0EmWJ').
1579100760: Sending CONNACK to 10343760ecfabc343234 (0, 0)
1579100760: Received SUBSCRIBE from 10343760ecfabc343234
1579100760:     smart/device/in/10343760ecfabc343234 (QoS 0)
1579100760: 10343760ecfabc343234 0 smart/device/in/10343760ecfabc343234
1579100760: Sending SUBACK to 10343760ecfabc343234
1579100767: New connection from 127.0.0.1 on port 1883.
1579100767: New client connected from 127.0.0.1 as 79dd260e-2fec-4ac5-9371-a7398bc036bd (c1, k60).
1579100767: Sending CONNACK to 79dd260e-2fec-4ac5-9371-a7398bc036bd (0, 0)
1579100767: Received PUBLISH from 79dd260e-2fec-4ac5-9371-a7398bc036bd (d0, q0, r0, m0, 'smart/device/in/10343760ecfabc343234', ... (111 bytes))
1579100767: Sending PUBLISH to 10343760ecfabc343234 (d0, q0, r0, m0, 'smart/device/in/10343760ecfabc343234', ... (111 bytes))
1579100767: Received DISCONNECT from 79dd260e-2fec-4ac5-9371-a7398bc036bd
1579100767: Client 79dd260e-2fec-4ac5-9371-a7398bc036bd disconnected.
1579100812: Received PUBLISH from 10343760ecfabc343234 (d0, q0, r0, m0, 'smart/device/out/10343760ecfabc343234', ... (80 bytes))
1579100815: Received PUBLISH from 10343760ecfabc343234 (d0, q0, r0, m0, 'smart/device/out/10343760ecfabc343234', ... (81 bytes))
1579100904: Client 10343760ecfabc343234 has exceeded timeout, disconnecting.
1579100904: Socket error on client 10343760ecfabc343234, disconnecting.
`

please notice that at the end i personally terminated the script because time left was 13:43:11 !

Please let me know if you need anything else!

kueblc, 2020-01-15:
The one that doesn't respond is actually running the tuya-convert firmware. You can restart the process without putting the device into pairing mode and it should be able to pick up where it left off.

There's a lot of MAC addresses in your wifi log so I can't identify which is the other device.

kueblc, 2020-01-24:
Hi @jokerigno, any luck?

kueblc, 2020-01-25:
Just released v2.4.2. Let me know if you still have issues with the new release.

jokerigno, 2020-01-26:
Hi, sorry for the delay but I still have issue. 
I successfully frashed 1 device using v2.4.2. But no luck with a second one.
Everything is fine till the moment when I chose bin to flash. I receive error "FILE NOT FOUND".

> Getting Info from IoT-device
> VTRUST-FLASH 1.1
> (c) VTRUST GMBH https://www.vtrust.de/35c3/
> READ FLASH: http://10.42.42.42/backup
> ChipID: 3439b9
> MAC: EC:FA:BC:34:39:B9
> BootVersion: 7
> BootMode: normal
> FlashMode: 1M DOUT @ 40MHz
> FlashChipId: 144068
> FlashChipRealSize: 1024K
> Active Userspace: user2 0x81000
> ======================================================
> Ready to flash third party firmware!
> 
> For your convenience, the following firmware images are already included in this repository:
>   Tasmota v8.1.0.2 (wifiman)
>   ESPurna 1.13.5 (base)
> 
> You can also provide your own image by placing it in the /files directory
> Please ensure the firmware fits the device and includes the bootloader
> MAXIMUM SIZE IS 512KB
> 
> Available options:
>   0) return to stock
>   1) flash espurna.bin
>   2) flash tasmota.bin
>   3) flash tasmota-minimal (3).bin
>   q) quit; do nothing
> Please select 0-3: 3
> Are you sure you want to flash tasmota-minimal (3).bin? This is the point of no return [y/N] y
> Attempting to flash tasmota-minimal (3).bin, this may take a few seconds...
> File Not Found
> 
> HAVE FUN!
> ======================================================
> Do you want to flash another device? [y/N]

Any hint?

kueblc, 2020-01-26:
Sounds like your device is running an older version of the intermediate firmware. Try `curl http://10.42.42.42/flash3` from another terminal when you get to this point.

jokerigno, 2020-02-05:
I've done it. Something changed but the issue persist.

Now this is what I see:

```
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
.Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
............
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N]
```
The device is plugged of corse but if I push the button nothing happens. No light, no blink, nothing.

kueblc, 2020-02-05:
Try `curl http://10.42.42.42/undo` instead. This should bring you back to the stock firmware so you can try again from the beginning.

jokerigno, 2020-02-06:
I receive an error:

`curl: (7) Failed to connect to 10.42.42.42 port 80: No route to host
`

Moskito204, 2020-02-25:
Hi All, 
I bought some BW-SHP6 recently - 15A version.

I'm current not able to make them connect to the vtrust-flash Network. However I can connect my Smartphone to it.

I'm running ubuntu 19.10 with a AC51 Dongle as WiFi interface.
The behavior I have is the script always retrying to send the smartconfig packets, but never discovers the plugs. Plugs are in fast blinking - so EZ config mode. - Exactly the same as described 2 posts up.

No connections are seen on the WiFi Logs, besides the Smartphone.

Can you help me? How can I help you getting aditional intel?

kueblc, 2020-02-25:
@jokerigno have you retried with `start_flash` still running?

@Moskito204 did you check your other logs?

Moskito204, 2020-02-25:
Hi kueblc,

I look at them and I didn't catch anything strange, besides the error on main (Lina 106, if not mistaken) for the psk script, which I see it is something common throughout the issues. (no impact right) 
However as I mentioned, the problem looks related with the establishment of the wifi connection, which doesn't seam related with the AP since I can connect my smartphone.

Will traces help? Which kind of logs should I take a better look at? Could the root cause be an issue with the smart configuration packets? Or is this a different Hardware from the plugs? 
I'm actually using Ubuntu on a VM. Can 
this also introduce some entropy?

S474N, 2020-04-11:
Hi, have also BlitzWolf® BW-SHP6, with original firmware 1.0.3. Have same problem with:

> Resending SmartConfig Packets
> .................
> SmartConfig complete.

Again, and again. 

My logs are here:
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4464571/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4464572/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4464574/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4464575/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4464576/smarthack-wifi.log)

S474N, 2020-04-11:
Another attempt
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4464595/smarthack-web.log)

S474N, 2020-09-05:
Tested with another PC (which connect to WiFi AP of RPi), about 100x, and then 3x BlitzWolf® BW-SHP6 with fw 1.0.3 successfully flashed :)