ct-Open-Source/tuya-convert #504: Clarification on expected wifi AP behavior

Virtual-Nomad, 2020-01-15:
Firstly my apologies as this may not be the right place to ask the question, however, I just wanted to confirm that my thinking is correct that despite the WiFi AP appearing to be configured and working if my phone refuses to join vtrust-flash (ie "the unable to join the network" ), I have a configuration issue and until that is resolved I won't be able to proceed.

[image]
[image]
[image]

kueblc, 2020-01-22:
It may be that there is a deeper issue, perhaps with IP assignments from `dnsmasq`.

Is your phone not receiving an IP address or is the problem earlier than that?

Your phone MAC address should appear in your `smarthack-wifi.log` as soon as a connection is attempted.

Note that you will not have a connection to the internet from `vtrust-flash`, this is expected.

kueblc, 2020-01-24:
Let me know if you need any further assistance, and good luck!

closed 2020-01-24 by kueblc