ct-Open-Source/tuya-convert #505: tuya-convert (docker) not working on arm devices + not working stop command.

Daviteusz, 2020-01-16:
Hello. Tuya-convert docker for arm devices is not working. 

```
pi@raspberrypi:~ $ git clone https://github.com/ct-Open-Source/tuya-convert
Cloning into 'tuya-convert'...
remote: Enumerating objects: 132, done.
remote: Counting objects: 100% (132/132), done.
remote: Compressing objects: 100% (90/90), done.
remote: Total 1240 (delta 74), reused 81 (delta 42), pack-reused 1108
Receiving objects: 100% (1240/1240), 3.68 MiB | 3.43 MiB/s, done.
Resolving deltas: 100% (769/769), done.
pi@raspberrypi:~ $ cd tuya-convert
pi@raspberrypi:~/tuya-convert $ docker build -t tuya:latest .
Sending build context to Docker daemon  5.413MB
Step 1/7 : FROM phusion/baseimage:0.11
0.11: Pulling from phusion/baseimage
c64513b74145: Pull complete
01b8b12bad90: Pull complete
c5d85cf7a05f: Pull complete
b6b268720157: Pull complete
e12192999ff1: Pull complete
d39ece66b667: Pull complete
65599be66378: Pull complete
Digest: sha256:6ba22a31a809de669243b79f982f7b2df733ead1ec45becff401b4d5d79150bb
Status: Downloaded newer image for phusion/baseimage:0.11
 ---> 14e6c478b00a
Step 2/7 : RUN apt-get update && apt-get install -y sudo iproute2 iputils-ping
 ---> Running in d9b0d2d0e7b0
standard_init_linux.go:211: exec user process caused "exec format error"
The command '/bin/sh -c apt-get update && apt-get install -y sudo iproute2 iputi                        ls-ping' returned a non-zero code: 1
pi@raspberrypi:~/tuya-convert $

```
After editing line "FROM" in "dockerfile" to " FROM phusion/baseimage:latest-arm", logs looks like this:

```
pi@raspberrypi:~/tuya-convert $ docker build -t tuya:latest .
Sending build context to Docker daemon  5.413MB
Step 1/7 : FROM phusion/baseimage:latest-arm
latest-arm: Pulling from phusion/baseimage
642790bfe707: Pull complete
2d7f0a2a454e: Pull complete
040caec04e29: Pull complete
ccc997cc74a3: Pull complete
86f8e17a2b6f: Pull complete
8c551090df61: Pull complete
Digest: sha256:1074423bc2694b1371732067ff9f1e9a51f1fd9fe57325a28f240d93f91132a5
Status: Downloaded newer image for phusion/baseimage:latest-arm
 ---> b4e54fb0e867
Step 2/7 : RUN apt-get update && apt-get install -y sudo iproute2 iputils-ping
 ---> Running in 7b1ca5b48c06
Get:1 http://ports.ubuntu.com/ubuntu-ports bionic InRelease [242 kB]
Get:2 http://ports.ubuntu.com/ubuntu-ports bionic-updates InRelease [88.7 kB]
Get:3 http://ports.ubuntu.com/ubuntu-ports bionic-backports InRelease [74.6 kB]
. . . 
. . .
Successfully installed paho-mqtt-1.5.0 pycryptodomex-3.9.4 sslpsk-1.0.0 tornado-6.0.3
Ready to start upgrade
Removing intermediate container e25018485a50
 ---> 415c7fa39a31
Step 7/7 : RUN mkdir -p /etc/service/tuya && cd /etc/service/tuya && ln -s /usr/bin/config.sh run
 ---> Running in 38c3dd871b8b
Removing intermediate container 38c3dd871b8b
 ---> b9912119db51
Successfully built b9912119db51
Successfully tagged tuya:latest
pi@raspberrypi:~/tuya-convert $

```
And it's working. I think "dockerfile" must contains image witch multi-arch support, like ubuntu:latest. Ubuntu is not good for docker using, but for tuya-convert it's working without problems.

Another issue. Command to stop "tuya-convert" in "docker" (docker-compose exec tuya stop) aren't working, because stop file contains:

```
#!/bin/bash
cd /usr/bin/tuya-convert
./stop_flash.sh
```

stop_flash.sh is removed from repository and it's integrated to start_flash.sh

ijl20, 2020-01-18:
I can confirm the default usage of tuya-convert 2.4.1 (i.e. install pre-reqs and run ./start_flash.sh) fails on a Raspberry Pi 3B+ while previous tuya-convert version 2.3.0 continues to work fine. The start_flash.sh script sets up the vtrust-flash lan ok, appears to recognize the smartplug ok (I tested several, e.g. 2Nice UP111), but doesn't load the (tasmota) firmware.

Temporary fix is to run the November 2019 2.3.0 tuya-convert, and I copied over the tasmota.bin (8.1.something) from  tuya-convert version 2.4.1., and I was able to flash 16 smartplugs of 3 types with 2.3.0 but could do none with 2.4.1 (I have a semi-permanent tasmota flash setup)

(BTW it would help to put the version somewhere in the code - I inferred it from github)

```
pi@raspberrypi:~$ uname -a
Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux

```
Log files from 2.4.1 here :
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4080170/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4080171/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4080172/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4080173/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4080174/smarthack-wifi.log)

kyounger, 2020-01-20:
Same issue. RPi4 here.

Linux raspberrypi 4.19.93-v7l+ #1290 SMP Fri Jan 10 16:45:11 GMT 2020 armv7l GNU/Linux

HEAD at tag `v.2.4.1`.

Daviteusz, 2020-01-24:
If you can't wait, use dev branch for arm devices.

git clone -b development https://github.com/ct-Open-Source/tuya-convert

closed 2020-01-24 by Daviteusz