ct-Open-Source/tuya-convert #509: Can't get this working. Don't know where to start... :/

xydix, 2020-01-20:
Hi!
I try to flash a Deltaco plug.
I run Ubuntu 18 on an Intel NUC.
This is how it looks like when I run ./start_flash.sh
```
ubuntu@ubuntu-desktop:~/tuya-convert$ ./start_flash.sh 
Checking for network interface wlp1s0... Found.
Checking UDP port 53... ss: invalid option -- 'H'
Usage: ss [ OPTIONS ]
       ss [ OPTIONS ] [ FILTER ]
   -h, --help          this message
   -V, --version       output version information
   -n, --numeric       don't resolve service names
   -r, --resolve       resolve host names
   -a, --all           display all sockets
   -l, --listening     display listening sockets
   -o, --options       show timer information
   -e, --extended      show detailed socket information
   -m, --memory        show socket memory usage
   -p, --processes     show process using socket
   -i, --info          show internal TCP information
   -s, --summary       show socket usage summary
   -b, --bpf           show bpf filter socket information
   -E, --events        continually display sockets as they are destroyed
   -Z, --context       display process SELinux security contexts
   -z, --contexts      display process and socket SELinux security contexts
   -N, --net           switch to the specified network namespace name

   -4, --ipv4          display only IP version 4 sockets
   -6, --ipv6          display only IP version 6 sockets
   -0, --packet        display PACKET sockets
   -t, --tcp           display only TCP sockets
   -u, --udp           display only UDP sockets
   -d, --dccp          display only DCCP sockets
   -w, --raw           display only RAW sockets
   -x, --unix          display only Unix domain sockets
   -f, --family=FAMILY display sockets of type FAMILY

   -A, --query=QUERY, --socket=QUERY
       QUERY := {all|inet|tcp|udp|raw|unix|unix_dgram|unix_stream|unix_seqpacket|packet|netlink}[,QUERY]

   -D, --diag=FILE     Dump raw information about TCP sockets to FILE
   -F, --filter=FILE   read filter information from FILE
       FILTER := [ state STATE-FILTER ] [ EXPRESSION ]
       STATE-FILTER := {all|connected|synchronized|bucket|big|TCP-STATES}
         TCP-STATES := {established|syn-sent|syn-recv|fin-wait-{1,2}|time-wait|closed|close-wait|last-ack|listen|closing}
          connected := {established|syn-sent|syn-recv|fin-wait-{1,2}|time-wait|close-wait|last-ack|closing}
       synchronized := {established|syn-recv|fin-wait-{1,2}|time-wait|close-wait|last-ack|closing}
             bucket := {syn-recv|time-wait}
                big := {established|syn-sent|fin-wait-{1,2}|closed|close-wait|last-ack|listen|closing}
Available.
Checking UDP port 67... ss: invalid option -- 'H'
Usage: ss [ OPTIONS ]
       ss [ OPTIONS ] [ FILTER ]
   -h, --help          this message
   -V, --version       output version information
   -n, --numeric       don't resolve service names
   -r, --resolve       resolve host names
   -a, --all           display all sockets
   -l, --listening     display listening sockets
   -o, --options       show timer information
   -e, --extended      show detailed socket information
   -m, --memory        show socket memory usage
   -p, --processes     show process using socket
   -i, --info          show internal TCP information
   -s, --summary       show socket usage summary
   -b, --bpf           show bpf filter socket information
   -E, --events        continually display sockets as they are destroyed
   -Z, --context       display process SELinux security contexts
   -z, --contexts      display process and socket SELinux security contexts
[... 641 lines cut ...]
Listening on 10.42.42.1:80
^[[32m[I 200120 21:51:37 web:2246]^[[m^O 302 GET /generate_204 (10.42.42.31) 1.95ms
^[[32m[I 200120 21:51:37 web:2246]^[[m^O 302 GET /generate_204 (10.42.42.31) 1.28ms
Listening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
wlp1s0: AP-STA-CONNECTED ca:d6:b4:9a:56:43
wlp1s0: AP-STA-CONNECTED 30:07:4d:f2:67:ec
wlp1s0: AP-STA-DISCONNECTED ca:d6:b4:9a:56:43
wlp1s0: AP-STA-CONNECTED ca:d6:b4:9a:56:43
^[[32m[I 200120 21:52:39 web:2246]^[[m^O 302 GET /generate_204 (10.42.42.31) 1.41ms
^[[32m[I 200120 21:52:39 web:2246]^[[m^O 302 GET /generate_204 (10.42.42.31) 1.15ms
Station 8c:b8:4a:44:aa:df trying to disassociate, but it is not associated
^C^CReceived SIGINT, exiting...
wlp1s0: interface state ENABLED->DISABLED
wlp1s0: AP-STA-DISCONNECTED ca:d6:b4:9a:56:43
wlp1s0: AP-STA-DISCONNECTED 30:07:4d:f2:67:ec
wlp1s0: AP-STA-DISCONNECTED d8:f1:5b:d1:f4:d8
wlp1s0: AP-DISABLED
nl80211: deinit ifname=wlp1s0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
Restarting NetworkManager...
```
I hope someone will try to help me somehow.
Thanks.

kueblc, 2020-01-22:
Looks like your version of `ss` doesn't support the `-H` argument (suppress headers). Can you provide the output of `ss -V` to get version information?

Additionally it looks like there are some issues with your `psk-frontend`. Is this a clean install of `tuya-convert` or have you updated from a previous installation? Have you run `install_prereq` since?

xydix, 2020-01-22:
This is a fresh install of ubuntu 16.04.6 LTS (Sorry, wrong in my first post. I thought it was Ubuntu 18).
Only installed tuya convert once, three days ago. It is a clean install.
ss -V gives ``` ss utility, iproute2-ss151103```

running ./install_prereq.sh gives.
```
ubuntu@ubuntu-desktop:~/tuya-convert$ ./install_prereq.sh 
Bra:1 http://se.archive.ubuntu.com/ubuntu xenial InRelease
Bra:2 http://se.archive.ubuntu.com/ubuntu xenial-updates InRelease        
Bra:3 http://se.archive.ubuntu.com/ubuntu xenial-backports InRelease      
Läs:4 http://security.ubuntu.com/ubuntu xenial-security InRelease [109 kB]
Läs:5 http://security.ubuntu.com/ubuntu xenial-security/main amd64 Packages [810 kB]
Läs:6 http://security.ubuntu.com/ubuntu xenial-security/main i386 Packages [631 kB]                                                                                              
Läs:7 http://security.ubuntu.com/ubuntu xenial-security/main Translation-en [310 kB]                                                                                             
Läs:8 http://security.ubuntu.com/ubuntu xenial-security/universe amd64 Packages [476 kB]                                                                                         
Läs:9 http://security.ubuntu.com/ubuntu xenial-security/universe i386 Packages [411 kB]                                                                                          
Läs:10 http://security.ubuntu.com/ubuntu xenial-security/universe Translation-en [195 kB]                                                                                        
Hämtade 2 843 kB på 31s (91,2 kB/s)                                                                                                                                              
Läser paketlistor… Färdig
Läser paketlistor… Färdig
Bygger beroendeträd       
Läser tillståndsinformation… Färdig
build-essential is already the newest version (12.1ubuntu2).
iw is already the newest version (3.17-1).
net-tools is already the newest version (1.60-26ubuntu1).
python3-dev is already the newest version (3.5.1-3).
python3-setuptools is already the newest version (20.7.0-1).
screen is already the newest version (4.3.1-2build1).
haveged is already the newest version (1.9.1-3).
python3-wheel is already the newest version (0.29.0-1).
curl is already the newest version (7.47.0-1ubuntu2.14).
git is already the newest version (1:2.7.4-0ubuntu1.7).
libssl-dev is already the newest version (1.0.2g-1ubuntu4.15).
dnsmasq is already the newest version (2.75-1ubuntu0.16.04.5).
hostapd is already the newest version (1:2.4-0ubuntu6.6).
mosquitto is already the newest version (1.4.8-1ubuntu0.16.04.7).
python3-pip is already the newest version (8.1.1-2ubuntu0.4).
Följande paket har installerats automatiskt och är inte längre nödvändiga:
  libllvm4.0 snapd-login-service
Använd ”sudo apt autoremove” för att ta bort dem.
0 att uppgradera, 0 att nyinstallera, 0 att ta bort och 9 att inte uppgradera.
Collecting git+https://github.com/drbild/sslpsk.git@use-byte-string-for-identity-hints
  Cloning https://github.com/drbild/sslpsk.git (to use-byte-string-for-identity-hints) to /tmp/pip-pttypx_r-build
  Requirement already satisfied (use --upgrade to upgrade): sslpsk===1.0.0 from git+https://github.com/drbild/sslpsk.git@use-byte-string-for-identity-hints in /usr/local/lib/python3.5/dist-packages
Requirement already satisfied (use --upgrade to upgrade): paho-mqtt in /usr/local/lib/python3.5/dist-packages
Requirement already satisfied (use --upgrade to upgrade): tornado in /usr/local/lib/python3.5/dist-packages
Requirement already satisfied (use --upgrade to upgrade): pycryptodomex in /usr/local/lib/python3.5/dist-packages
You are using pip version 8.1.1, however version 19.3.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
Ready to start upgrade
```
Checking pip version.
```
ubuntu@ubuntu-desktop:~/tuya-convert$ pip -V
pip 19.3.1 from /home/ubuntu/.local/lib/python3.5/site-packages/pip (python 3.5)
ubuntu@ubuntu-desktop:~/tuya-convert$ pip2 -V
pip 19.3.1 from /home/ubuntu/.local/lib/python2.7/site-packages/pip (python 2.7)
ubuntu@ubuntu-desktop:~/tuya-convert$
```

u2n, 2020-01-22:
Think I can help here. The version of ss installed will not have the -H flag, which was introduced in a later release. Didn't try enabling backports myself (not a sure fix and could introduce other problems), but did a fresh install of 18.04 and everything worked.

xydix, 2020-01-23:
I did an upgrade on my ubuntu. Now it is working. Thank you guys. Really awesome software.

closed 2020-01-23 by xydix