ct-Open-Source/tuya-convert #510: Successful  with Teckin Bulb before (same batch), now not working

uatuba, 2020-01-20:
Name: Teckin RGBW Bulb
Firmware Verision: 1.0.5

I bought several of these bulbs about a year ago. I installed them in various hard to reach locations, so I've flashed them as I have been able to get to them (the most recent successful flash was about a month ago). I had to reinstall my Pi OS due to an SD card issue, so I only have the latest version of tuya-convert. Now two of these bulbs from the same batch--having been connected to Tuya until today--will not successfully flash.

Here are my logs:

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4087915/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4087916/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4087917/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4087918/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4087919/smarthack-wifi.log)

Any help is appreciated as always.

kueblc, 2020-01-22:
Try
```
sudo -H python3 -m pip install --reinstall git+https://github.com/drbild/sslpsk.git@use-byte-string-for-identity-hints
```

uatuba, 2020-01-24:
This didn't work. Here are the current logs if they'll help:

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4106816/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4106817/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4106818/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4106819/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4106820/smarthack-wifi.log)

kueblc, 2020-01-24:
The PSK error you were experiencing earlier looks to be resolved.

It looks like the device is reporting a successful upgrade command
```
POST /gw.json?a=s.gw.upgrade.updatestatus&gwId=08063072b4e62d2243bb&t=1579579968&sign=ba691b5ed523314daaf6f67235920469
Host: 10.42.42.1
Ty-Ua: HW
Content-Length: 69
Connection: keep-alive
Content-Type: application/x-www-form-urlencoded; charset=UTF-8

payload 551D445021C8736C2BA2259FEEFE501EE99EC8C6942F8C8403A6B894C9383C7D
Answer s.gw.upgrade.updatestatus
reply {"e":false,"success":true,"t":1579579968}
[32m[I 200120 22:12:48 web:2246][m 200 POST /gw.json?a=s.gw.upgrade.updatestatus&gwId=08063072b4e62d2243bb&t=1579579968&sign=ba691b5ed523314daaf6f67235920469 (10.42.42.22) 5.03ms
^CReceived SIGINT, exiting...
```
But it is immediately followed by the script exiting. Perhaps letting the script run longer will allow it to run to completion.

kueblc, 2020-01-24:
I had a little difficulty determining the device in question, hopefully I am looking at the right one. If not you can let me know the `gwId`.

I recommend moving the logs to your backup folder so that your next attempt will start with clean logs.

uatuba, 2020-01-25:
It never would install the firmware, but rather just keep trying. I deleted the entire tuya-convert directory, rebooted, and started the process from install to flash again. This time it worked. Not sure why, but it's working now.

closed 2020-01-25 by uatuba