ct-Open-Source/tuya-convert #511: No screen session found.

neurax6, 2020-01-20:
On a fresh up2date kali , prereq script run without error. 
Getting same result with a a fresh Raspbian lite image.
root@kali:~/Downloads/tuya-convert# ./start_flash.sh

================
"Warning text"
================

Cleaning up...
No screen session found.
No screen session found.
No screen session found.
No screen session found.
Closing AP
Exiting...

root@kali:~/Downloads/tuya-convert# screen -v
Screen version 4.07.00 (GNU) 02-Oct-19

root@kali:~/Downloads/tuya-convert# uname -a
Linux kali 4.19.93-Re4son-v7l+ #1 SMP Wed Jan 8 08:35:42 AEDT 2020 armv7l GNU/Linux

root01, 2020-01-21:
Same problem

kueblc, 2020-01-22:
Please post your logs

neurax6, 2020-01-22:
What log do you want ? I’ve posted the result of ./start_flash.sh
I check syslog and find nothing particular.
If you want to see something specifics…

[cid:image003.jpg@01D5D12C.074905D0]

kueblc, 2020-01-22:
Check inside `scripts` and `backup` within your `tuya-convert` directory. You should see `smarthack-*.log` files.

neurax6, 2020-01-22:
Nothing , no backup directory and no logs files after running the ./script

kueblc, 2020-01-22:
I am not really sure how to help then. "Warning text" does not occur in our codebase, and "Cleaning up..." happens when the script exits. It must be exiting early from some error, but I have no way of knowing what that error might be.

neurax6, 2020-01-22:
Screen session doesn’t seem to launch  normally.  No screen session in the htop when I run the script but a bash while seem to continue running after the script exits.

Can you give me a list of manual command to try. Or a bash script with stop/debug point , im not familliard with bash script debugging(for now).

For now i will setup a VM with usb wifi dongle and try  to reproduce this bug on a x86.

[cid:image001.jpg@01D5D13D.207C87A0]

neurax6, 2020-01-22:
Forget about the bash script still running, don’t came from  not the same process.

kueblc, 2020-01-24:
You say that `start_flash` produced
```
================
"Warning text"
================
```
But this is not in our scripts. Can you please explain whether you modified the script or abridged the log? This is important in determining what part of the script is causing the error.

See https://stackoverflow.com/questions/9080431/how-execute-bash-script-line-by-line for an example of how you might debug a bash script.

neurax6, 2020-01-24:
I’ve removed the warning message to shorten my post. I did not modify the script in any way.

Thanks for the links , I msg you if I got any new clues.

sotosvar, 2020-01-25:
After you run the script [./start_flash.sh] in the next step try to write [yes + enter] not [y+enter]
I thing this is your problem.
I also had  the same problem too :-)

kueblc, 2020-01-26:
Let me know if you need more help

closed 2020-01-26 by kueblc

udondan, 2020-01-26:
The actual error message is a bit hard to see. It appears to be part of the warning text. I did not see it at first and landed in this issue when googling for `No screen session found.`

dmstr82U2, 2020-01-30:
> After you run the script [./start_flash.sh] in the next step try to write [yes + enter] not [y+enter]
> I thing this is your problem.
> I also had the same problem too :-)

dmstr82U2, 2020-01-30:
======================================================
PLEASE READ THIS CAREFULLY!
======================================================
TUYA-CONVERT creates a fake update server environment for ESP8266/85 based tuya devices. It enables you to backup your devices firmware and upload an alternative one (e.g. ESPEasy, Tasmota, Espurna) without the need to open the device and solder a serial connection (OTA, Over-the-air).
Please make sure that you understand the consequences of flashing an alternative firmware, since you might lose functionality!

Flashing an alternative firmware can cause unexpected device behavior and/or render the device unusable. Be aware that you do use this software at YOUR OWN RISK! Please acknowledge that VTRUST and c't Magazine (or Heise Medien GmbH & Co. KG) CAN NOT be held accountable for ANY DAMAGE or LOSS OF FUNCTIONALITY by typing yes + Enter

======================================================
Cleaning up...
No screen session found.
No screen session found.
No screen session found.
No screen session found.
Closing AP
Exiting...

I am getting this issue both in my Proxmox VM and also, installed it on a Raspberry Pi 3B+, I've turned off my "Adguard" to make sure something wasn't being blocked, still the issue persists. Any assistance and/or direction will be appreciated.
P.S @kueblc I checked for the smarthack-*.log files, they didn't exist.

kueblc, 2020-01-30:
I'm not sure I follow, did you "type `yes` + `Enter`"? You can ignore the `No screen session found`, it just means the services never started in the first place.

udondan, 2020-01-30:
> I checked for the smarthack-*.log files, they didn't exist.

I noticed the same thing. Not a single log was written. No errors in the output. That was on CentOS 7. 

I then switched to docker. Same config, logs are created. 👍 Though a bit tricky to get to the logs then. Need to mount the scripts dir, so you basically need to store the scripts outside the container, which is a bit strange. Would be better if logs were written to another or sub dir.

dmstr82U2, 2020-01-30:
I'm an idiot, I was just hitting ”enter” as opposed typing ” yes” and then
enter. I guess the old adage ” reading is fundamental”! 🙄

Thanks for your help.

Rob190, 2024-02-05:
I had this exact same problem. The screen I am using is quite small and it is difficult to distinguish between 'yes + Enter' and 'yes / Enter' which is what I thought it said. Can you just use the standard y/N here please?