ct-Open-Source/tuya-convert #513: LSC Smart Connect RGBW E27 Blub flashing doesn't work

root01, 2020-01-22:
Hello,
I'm trying to flatten this bulb but it's not working. The LSC Smart Connect MR16 380lm RGBW Bulb flashed without problems but the E27 stays at 

"SmartConfig complete.
Resending SmartConfig Packages
............"

Hang up then breaks off.
Is there anything else I could try?

kueblc, 2020-01-22:
I don't believe this device is supported, as it does not use an ESP. To be sure you would need to share your logs as indicated in the instructions.

root01, 2020-01-22:
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4099357/smarthack-wifi.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4099358/smarthack-udp.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4099359/smarthack-mqtt.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4099360/smarthack-web.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4099361/smarthack-psk.log)

root01, 2020-01-22:
The blub stops short blinking while the flashing process. something happens

kueblc, 2020-01-22:
Unfortunately not an ESP, as revealed by the MAC address in your wifi log

closed 2020-01-22 by kueblc

root01, 2020-01-22:
Ok. that's bad. Thanks