ct-Open-Source/tuya-convert #515: Outdated SSL May Brick and Clarity in Documentation

u2n, 2020-01-23:
**Part 1:** If execution reaches the secure handshake point with an outdated SSL library onboard, apparently enough data is flashed to brick [the target](https://templates.blakadder.com/esicoo.html), even though the overall flash fails with error "could not establish sslpsk socket."

Unfortunately only notes and no logs as this happened ~5 weeks ago and no idea anything wrong.

Laptop was running 16.04 Ubuntu and, since start_flash would not complete, I was debugging by stepping through the code one line at a time. After the third fix (outdated ss), I let 'er rip. The out-of-date revelation and bricking apparently occurred simultaneously.

Updated to 18.04 this AM and prepared to continue from stop point but found the device unresponsive. The LED was on but it could not be made to enter fast-blink (or any blink) pairing mode. Finally, holding the button down, at about 15 seconds the LED turned off (never supposed to happen) and the internal relay toggled. No amount of anything tried has brought it back from the dead.

A version check on the SSL library might avoid this. Or maybe just a list of minimum release numbers and let users check manually. It appears that many issues are due to old software.

**Part 2:** May come down to operator error/documentation. Tossing the first plug, went for No. 2.

18.04 update ran without error. From home page here, requirements: first flash binary < 512 kB and include bootloader. So copied tasmota-lite.bin from [the release page](https://github.com/arendst/Tasmota/releases) thinking it was complete firmware: "downloads have been compiled with ESP8266." Picked it at flash menu and everything seemed to work. Entered wi-fi credentials, saved, device rebooted, and... It's stuck in slow-blink mode with no wi-fi.

So nailing down which files to flash (and in which order) before No. 3. Learning how _not_ to do it along the way.

**Great project**, BTW, wishing continued success. (And condolences @kueblc on your recent loss.)

kueblc, 2020-01-23:
You don't still have the first plug? It may not be bricked, those sound like symptoms of the device running the intermediate firmware, but not connecting to the AP in time. [We may be able to recover it still](https://github.com/ct-Open-Source/tuya-convert/wiki/Troubleshooting).

For the second plug, it sounds like the flashing process completed successfully, or else you would not be able to get to the wifi config page. Have you triple checked your router's listing for the device? Or perhaps you [inadvertently entered the wrong credentials](https://tasmota.github.io/docs/#/help/Device-Recovery)? In any case, these are Tasmota issues to debug.

For Tasmota I recommend doing the initial flash with the bundled version, [issuing `reset 5`](https://tasmota.github.io/docs/#/Tuya-Convert), and then using the upgrade mechanism to install your desired version.

u2n, 2020-01-23:
Have both and rechecked, but this time not with two cell phones and a laptop as before. Positioned devices near router and used Site Survey in DD-WRT. Amazingly (to me at least), found signs of life.  Update forthcoming...

Reviewing docs now, I see timing as much of the problem. Returning to a task like this after too long should compel a more thorough review before proceeding.

u2n, 2020-01-24:
Thanks, @kueblc for recovery link; [fast power cycle](https://tasmota.github.io/docs/#/help/Device-Recovery?id=fast-power-cycle-device-recovery) worked (except needed more than 4x plug/unplug cycles to activate; 5x-8x did it). Then the Part 2 device was essentially done, as you noted, just connected to its AP, keyed the local router credentials and hit Save. After auto-reboot, it joined the LAN and showed Tasmota GUI.

The Part 1 device was a recovery and re-flash. It connected to the tuya-convert server ok, but would not accept the upgrade (tried twice) until after a re-stock. Then flashed the same -lite file (which _is_ a complete firmware), immediately connected to its AP and did a rinse and repeat. All well with both units, and flashed a third the same way (minus the freakout).

So closing this now since devices are recoverable. Takeaways are:
- Flashing directly to working firmware *can* be a timesaver.
- Do update system before flash
- Key router credentials immediately
- Don't fear the brick: This method is sound, device firmware is robust, and recovery is there if needed.

This is now true home automation. Whether ultimately going with MQTT, direct commands or some other control scheme, not sure yet. But in any case, **thanks to the people here, and all the devs and contributors.** Hopefully Tuya sees that their devices are more appealing when consumers are free to choose a preferred interface. Freedom is a good thing.

kueblc, 2020-01-24:
Excellent @u2n, that's great to hear! I am glad you found our software useful, you are very welcome.

closed 2020-01-24 by kueblc