ct-Open-Source/tuya-convert #516: Device did not appear with the intermediate firmware error

abhitheawesomecoder, 2020-01-23:
I am having receiving the 'Device did not appear with the intermediate firmware' error while attempting to flash smart breaker switch with model no scw nf101. 

https://www.aliexpress.com/item/32961129650.html
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4102000/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4102001/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4102002/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4102004/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4102005/smarthack-wifi.log)

Attached are my logs. I appreciate the assistance.

kueblc, 2020-01-23:
Looks like the device successfully grabbed the intermediate firmware.
- Restart `start_flash`
- When you get to the pairing prompt, plug your device back in
- Do not connect your phone or put your device into pairing mode, just press enter
- With any luck the script will pick up where it left off

DirkTas67, 2020-01-25:
Hello,

Have recently bought 2 novostella floodlights, they seems to have a new Model number -> UT88836 instead of the UT88835, I have the same problem and retrying doesn't help.

GeorgeIoak, 2020-01-25:
Not sure if it helps but I just did 2 of these lights about 1-2 weeks ago. I'm not sure of the actual model number since nothing is printed on the lamp itself. I do remember that on 1 of them it seemed dead during the process but when I unplugged and plugged the device back in the process continued and I was able to successfully flash the light.

DirkTas67, 2020-01-25:
Well, Thank you !!! I'm glad I posted my problem here.

Until now I only tried to flash the first device several times but it didn't work.
After your message I decided to try the second one and it worked from the first time, after that I tried the first device and it needed 5 total retry's to get converted!

kueblc, 2020-01-25:
Timing can be a bit hit or miss. Glad you sorted it out! I'll be here if you run into any more issues.

closed 2020-01-25 by kueblc

sneezy79, 2020-02-02:
How did you get to the logs.  Unfortunately, I'm very new with Linux...