ct-Open-Source/tuya-convert #518: start_flash.sh fails if ufw is enabled

ndfred, 2020-01-24:
`start_flash.sh` does a great job of disabling services that run on the various ports it uses, but doesn't disable ufw (infinite loop in `Resending SmartConfig Packets`). After I disabled ufw and ran the script again everything worked. Maybe the script should check if the firewall is enabled, and suggest disabling it.

kueblc, 2020-01-24:
Great catch! I had noticed some other folks with firewall issues. #450 #464

Any suggestions for the best way to detect and temporarily disable firewalls?

kueblc, 2020-01-25:
Let me know what you think of this
```bash
check_firewall () {
	if sudo systemctl stop firewalld.service &>/dev/null; then
		echo "Attempting to stop firewalld.service"
		echo "When done, enable with: sudo systemctl start firewalld.service"
	fi
	if sudo ufw status | grep -qw active; then
		sudo ufw disable
		echo "When done, enable with: sudo ufw enable"
	fi
}
```

commit 52279571 (2020-01-29): Attempt to stop firewalls, fix #518
PR #524 (2020-01-29): Release v2.4.3
- Update `sslpsk` installation
- Add timeout on backup and allow user to continue anyway, fix #437 
- Attempt to stop firewalls, fix #518
closed 2020-01-29 by kueblc