ct-Open-Source/tuya-convert #519: PSK front end crashes when trying to flash Gosund SP111

barbudor, 2020-01-24:
Hello
This is my 2nd set of 4x Gosund SP111 sockets. 1st set has been converted to tasmota using tuya-convert on Jan 5th without any issue using the same context as below.
I am facing issues now with this 2nd set.
I have openned the sockets and can confirm that they are identical to my 1st set and based on ESP82xx.

- Environment : laptop running Lubuntu 19.10 LIVE 
- apt update / apt upgrade performed before cloning tuya-convert
- instal_prereq.sh performed
- edit config.txt and change WLAN interface => wlp1s0
- start_flash
- connect smartphone to vtrust-flash (=> 10.42.42.99)
- power the socket, force pairing
- press ENTER

I can see the socket connecting to the wifi `wlp1s0: AP-STA-CONNECTED 24:62:ab:2f:7d:30`
And running a network scanner on the smartphone I can see the IP assigned to the socket

I get the continuous messages regarding sending smart config packets but nothing else happens

Looking at the log I noticed that PSK front-end crashed as below
```Traceback (most recent call last):
  File "./psk-frontend.py", line 111, in <module>
    main()
  File "./psk-frontend.py", line 107, in main
    p.data_ready_cb(s)
  File "./psk-frontend.py", line 78, in data_ready_cb
    self.new_client(_s)
  File "./psk-frontend.py", line 66, in new_client
    hint=self.hint)
  File "/usr/local/lib/python3.7/dist-packages/sslpsk/sslpsk.py", line 110, in wrap_socket
    sock.do_handshake()
  File "/usr/lib/python3.7/ssl.py", line 1139, in do_handshake
    self._sslobj.do_handshake()
OSError: [Errno 0] Error
```

All logs available on the following gist : https://gist.github.com/barbudor/f842045888dee6880a74654d36b3d5bf

I tried again and again .. WIth some luck I was able to flash 2 of the sockets. But with the 3rd one it was impossible to reproduce the good sequence of powering-up the sequence and pressing enter.
I forgot to save the logs for the 2 successfull flashes so I don't know if the same issue was present. The PSK logs from my successful flashes back on Jan 5th do not have the issue.

Any suggestion would be appreciated.

Thanks a lot.

kueblc, 2020-01-25:
Would you try the `development` branch? You will need to run `install_prereq.sh` again in the event that the `sslpsk` library does not update.

barbudor, 2020-01-25:
:+1:
It all went smoothly with the dev branch.
Thanks a lot @kueblc for you work and support

closed 2020-01-25 by barbudor