ct-Open-Source/tuya-convert #522: Efun SH331W - trouble getting in/out of backup

MakerMatrix, 2020-01-26:
I bought a 4-pack of these.  I got two of them successfully flashed to Tasmoda.  The other two have a lot of trouble connecting to and staying connected to the wifi on my Pi.  I'm not 100% sure but I believe the problematic two were the first pair I was working with, during which time I was using my iPhone11 to connect to vtrust-flash, which I later recognized was bad because the iPhone would fall off the wifi every time I allowed the screen to go to sleep.  So, not sure if these modules suffered some damage from that, or if it's a red herring.  But as soon as I switched to another device for connecting to vtrust-flash, I got the other two modules flashed right away with nearly no hassle at all.

After that I went back to the other (first) two and tried too many things to list.  I did eventually get one of them into backup mode twice (by adjusting the timing of when I plugged it in), but it only got about 5k transferred and then the backup hung.

I'm attaching my logs for the last ~dozen attempts, and a tcpdump pcap of all the Pi's wlan0 traffic for the last attempt.  If you have any advice I'd appreciate it.

[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4113328/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4113329/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4113330/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4113331/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4113332/smarthack-mqtt.log)
[pcap.zip](https://github.com/ct-Open-Source/tuya-convert/files/4113338/pcap.zip)

eku, 2020-01-28:
I observe exactly the same with Gosund SP111. The backup hangs after about 128KB.

MakerMatrix, 2020-01-28:
Amazon delivered 2 more 4-packs yesterday and I successfully flashed all 8 of them.  I think I observed stalling at the beginning of backup (0-1% complete) 2-3 times total.  I was able to just start the script over and these flashed just fine.

I still have problems with the same two that I'd first tried to flash (when my phone was falling off the vtrust-flash wifi, midstream).  So I guess those two are somehow bricked and I'll need to figure out how to open them and do it over serial.

MakerMatrix, 2020-01-28:
> I observe exactly the same with Gosund SP111. The backup hangs after about 128KB.

In my experience this can also be caused by a timing issue.  On the Efun ones I've got, if you plug it in too early, the backup will hang by the time the script gets to it.  I find that plugging them in at the same moment that you hit ENTER to start the flash process was quite reliable.

kueblc, 2020-01-29:
Added an option to skip backup if it times out in the most recent release, v2.4.3. Let me know if you have any issues after updating. Good luck!

closed 2020-01-29 by kueblc