ct-Open-Source/tuya-convert #526: Unable to flash tuya bulb: fcc id: 2AJK8-LZ803

jimtino, 2020-01-30:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
_Please add it to the wiki instead!_

**Do you have a question about configuring your newly flashed device?**
_Please ask in your alternative firmware's support channel instead!_

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

**Please remember to close your issue when the problem has been addressed!**

jimtino, 2020-01-30:
I'm using tuya convert version v2.4.3 . and am getting the "Device did not appear with the intermediate firmware" I don't see anything onerous in the error logs. Can someone help me decipher what may be going on?

Thanks so much. Logs attached:
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4132312/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4132313/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4132314/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4132315/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4132316/smarthack-wifi.log)

kueblc, 2020-01-31:
Looks like the device is taking the intermediate firmware, but then restarting back into the stock firmware. Might suggest that the upgrade image is corrupt in some way.

jimtino, 2020-01-31:
Maybe I missed something. Where do I get the upgrade image? Is it part of tuya convert? I'm using a rasberry pi. Do I need to reimage the pi? Thanks so much.

kueblc, 2020-02-04:
The upgrade image (`upgrade.bin`) is included with tuya-convert. It might be that it was corrupted in download, or that your device is in some way incompatible with the current version. It is difficult to say with the limited information.

kueblc, 2020-05-12:
Closing due to age, please feel free to comment if this is still an issue

closed 2020-05-12 by kueblc