ct-Open-Source/tuya-convert #527: Issues creating AP with Tuya-convert (using Wiki based setup)

bugs181, 2020-01-31:
Using an almost identical setup to the Wiki, I'm having issues putting my device into AP mode.

The setup I'm trying to replicate is:
- Ubuntu 18.04.3 64Bit in VirtualBox on Win10 with a cheap RTL8188CU Wifi Adapter connected to the VM

The only difference here is that I'm running it on OSX with the Wifi adapter in pass-through mode.

**Using Ubuntu UI to create the hotspot works great. (The access point shows up and I'm able to connect to it.)**

However, using `start_flash.sh` I've had no such luck. There are no errors reported at all but the access point does **NOT** show up. I've also tried additional versions of `hostapd` with the RTL8188CU driver without luck.

Here's my exact setup:

`lsb_release -a`:

```
Distributor ID: Ubuntu
Description: Ubuntu 18.04.3 LTS
Release: 18.04
Codename: bionic
```

`iwconfig`:

```
wlx00e04d0b774f.  IEEE  802.11.  ESSID:off/any
      Mode:Managed. Access Point: Not-Associated. Tx-Power=20. dBm

lo.           no wireless extensions.
```

**I was hoping that whomever added this setup to the wiki could shed some light on how to go about setting this up for tuya-convert.**

kueblc, 2020-01-31:
@haniham via #417

haniham, 2020-02-01:
Well - i added the ID of the wifi adapter into the script 

Does it work to connect to other wifi networks from your VM?

bugs181, 2020-02-04:
> Well - i added the ID of the wifi adapter into the script
> 
> Does it work to connect to other wifi networks from your VM?

Yes it does. It can connect to wifi networks and using the Ubuntu UI can also create a hotspot. Just not from `start_flash.sh`.

kueblc, 2020-02-04:
@bugs181 are you able to determine what Ubuntu UI uses to create it's hotspot? We could borrow from them, or even just use the hotspot feature to create `vtrust-flash`. Then you would need to run `dnsmasq` manually, (see `setup_ap`) and the rest of the services (see `setup` in `start_flash`).

kueblc, 2020-02-04:
Alternatively, if you have [`create_ap`](https://github.com/oblique/create_ap) installed, you could try the `create-ap` fork.
```
git clone -b create-ap https://github.com/kueblc/tuya-convert.git
```

kueblc, 2020-05-12:
Closing due to age, please feel free to comment if this is still an issue

closed 2020-05-12 by kueblc

giuliandenicola1, 2020-07-04:
Hi I still have this issue

kueblc, 2020-07-04:
@giuliandenicola1 this is an old closed issue, you should create a new issue if you are not the original author. However in this case, your issue is written out clearly for you in your log, your wifi dongle does not support AP mode. You must use a different wifi card, there is nothing we can do to make your dongle support AP mode.

giuliandenicola1, 2020-07-05:
> @giuliandenicola1 this is an old closed issue, you should create a new issue if you are not the original author. However in this case, your issue is written out clearly for you in your log, your [...]

Well, I kinda did not fix this issue. Instead of using VMware/VirtualBox I installed it on the host pc with dual boot and now it's working. So it's definitely not the wifi dongle that doesn't work because it does.