ct-Open-Source/tuya-convert #528: tuya-convert not flashing Nooie LB127 Smart Led Bulb

aviogit, 2020-02-01:
Hi, I'm having trouble flashing a [Nooie LB127 Smart Led Bulb](https://www.amazon.com/Dimmable-Multicolor-Compatible-Required-Equivalent/dp/B07X1KYSBJ).

I tried both the dockerized version and the normal one, but the outcome is always the same:

> Device did not appear with the intermediate firmware
> Check the *.log files in the scripts folder
> Do you want to try flashing another device? [y/N] 

The only strange thing in the logfiles is the `NO_SHARED_CIPHER` repeated message in `smarthack-psk.log`:

> new client on port 443 from 10.42.42.31:38386
> could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:852)
> don't panic this is probably just your phone!

My phone is `64:a2:f9:f5:c1:c9 -> 10.42.42.31` while the Nooie bulb is `68:57:2d:6d:f1:d7`. Below you can find all the logfiles.

Thanks for any help.

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4143284/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4143285/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4143286/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4143287/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4143288/smarthack-wifi.log)

kueblc, 2020-02-01:
Do you have any firewalls? Check for `ufw` and `firewalld`

aviogit, 2020-02-01:
I never had any firewall issue on this pc, so I'd say no.

> #> sudo ufw status
> WARN: Duplicate profile 'Apache', using last found
> WARN: Duplicate profile 'Apache Secure', using last found
> WARN: Duplicate profile 'Apache Full', using last found
> Status: inactive

> #> firewalld
> 
> Command 'firewalld' not found, but can be installed with:
> 
> sudo apt install firewalld

I was looking at logfiles by other users... mine have mostly nothing in the web part. It's just the phone/tablet that visits the sign in page of the wifi. So strange...

aviogit, 2020-02-01:
Yep, if I netcat from my phone to 10.42.42.1, the phone IP address pops up in `smarthack-udp.log`, so I'd say this part works.

aviogit, 2020-02-01:
The other bulb does exactly the same. I was afraid that having used the first one for one day with the original app may have updated its firmware in the background.

aviogit, 2020-02-03:
I think I found the problem. `start_flash.sh` assumes that the device is at `10.42.42.42` (it's hardcoded in the script) but mine acquires its address via DHCP, so between `10.42.42.10` and `10.42.42.40`. In fact, it becomes `10.42.42.21` (see [attached arp table](https://github.com/ct-Open-Source/tuya-convert/files/4150348/tuya-nooie-arp.txt)).

I'll try to change the destination from `10.42.42.42` to `10.42.42.21` and I'll report back if it works.

[tuya-nooie.pcap.gz](https://github.com/ct-Open-Source/tuya-convert/files/4150361/tuya-nooie.pcap.gz)

aviogit, 2020-02-03:
Nope... that part comes later... so it's a multicast problem? The device doesn't subscribe to 226.120.89.84?

kueblc, 2020-02-03:
> `start_flash.sh` assumes that the device is at `10.42.42.42` (it's hardcoded in the script)

This is intentional and should not be changed, as you have likely discovered. The static IP `10.42.42.42` is only taken by the device once it is running our intermediate firmware, this is the signal it uses to notify the script of success and to proceed.

> The device doesn't subscribe to `226.120.89.84`?

Multicast is used by the smartconfig process to transmit AP credentials. No subscription is needed. The device enters sniffer mode and extracts information encoded in the IP addresses of the multicast packets. This is how it can transmit information before it is connected to the network.

There is nothing wrong with the smartconfig process as we can see the device in your wifi log. The problem is that there are no requests from the device in your psk log.

aviogit, 2020-02-04:
Making a comparison between a successful pairing via app:

[image]

and an unsuccessful pairing via `tuya-convert`:

[image]

I'd say I have some DNS problem...

I've already tried the pairing on a fresh raspberry, but the result is the same. I've also found a `region = "US"` in `main.py` in `smartconfig` directory, tomorrow I'll try to set it to EU.

kueblc, 2020-02-04:
> I'd say I have some DNS problem...

This was my original thought, but now it is looking like the device is trying to access an unusual port (4433). We don't have anything listening on this port which would explain why it isn't getting a response, and thus why the process is failing. This is what we must investigate.

It looks like it is expecting MQTT. Perhaps as a quick test you can edit `psk-frontend.py` so that MQTTS is answered on `4433` instead of `8886`.

> I've also found a `region = "US"` in `main.py` in `smartconfig` directory, tomorrow I'll try to set it to EU.

This shouldn't make a difference. Even if it could, like I said earlier, there is no issue with smartconfig since we can see the device connecting. This is the only role of smartconfig and the third connected device.

kueblc, 2020-02-04:
Ah, somehow I did not notice this before, but your device does not use an ESP82xx and therefore is not supported by tuya-convert, Tasmota, ESPHome, etc. This is apparent by the device's MAC address using an OUI lookup tool. Supported devices will have an OUI of Espressif Inc.

Better luck with your next device!

closed 2020-02-04 by kueblc

aviogit, 2020-02-04:
Oh, that's really bad news. Well, at least I'm done suffering :)