ct-Open-Source/tuya-convert #529: Flashing Teckin SP25 terminates while sending SmartConfig packets

muckngrind4, 2020-02-04:
I've tried flashing 2 of a 4 pack of Teckin SP25 smart plugs without success.  Here is the console output:

```
tuya-convert v2.4.3
Checking for network interface wlx74da380d4584... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
................/start_flash.sh: line 129:  4944 Terminated              ./smartconfig/main.py
........................................................................................................
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] n
======================================================
Cleaning up...
Closing AP
Exiting...

```

Some observations:
- As soon as I hit enter to start pairing the device changed from "quick mode" (fast blinking) to "AP mode" (slow blinking) and the "Terminated" error message appeared.
- No backup was performed.
- I was suspicious that Teckin perhaps changed out the ESP8266 chip so I checked the MAC address against an OUI lookup and it does return "Espressif Inc."
- Both devices are unresponsive--hopefully not bricked but might be.

Logs attached. 
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4151578/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4151579/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4151580/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4151581/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4151582/smarthack-wifi.log)

I'm not sure what to try next.  Any help is appreciated.

Thanks,
Derek

kueblc, 2020-02-04:
Looks like the device is probably running the intermediate firmware. You can restart the process, but you do not need to connect a phone or put the device in pairing mode this time. Just press enter and plug the device back in.

muckngrind4, 2020-02-04:
Wow.  So simple and totally worked. 

I saw that in another thread, tried it and it didn't work.  Clearly it was a user error because I just did it again to both devices and they are now properly flashed.  

Thank you very much!

closed 2020-02-04 by muckngrind4

CaptainSweatpants, 2021-03-05:
> Wow. So simple and totally worked.
> 
> I saw that in another thread, tried it and it didn't work. Clearly it was a user error because I just did it again to both devices and they are now properly flashed.
> 
> Thank you very much!

Could you post the pin outs for the SP25? I’ve been searching forever to find them, but nothing for the sp25 specifically. 

> Wow. So simple and totally worked.
> 
> I saw that in another thread, tried it and it didn't work. Clearly it was a user error because I just did it again to both devices and they are now properly flashed.
> 
> Thank you very much!

You wouldn’t happen to have (or know where I could find) the pin out references for the SP25 would you?

lambchop01, 2022-03-18:
Stumbled across this while looking for a solution my unresponsive plugs.  I'll have to try it again and see if that works!

I am also looking for the pinout references for a sp25...