ct-Open-Source/tuya-convert #530: Want add another Tasmoate Version bigger than 512kb

iqmend, 2020-02-04:
Hello,
i want add in tuya-convert/files an older tasmota version. But all versions are bigger than 512kb. 
Only the minimal and lite version are smaller but i dont want install this version.
Why do we have this limit and can i turn it up to 600kb ? Because than i can install every version i want.

[image]

[image]

kueblc, 2020-02-04:
Use the included Tasmota and then upgrade to the version you want

closed 2020-02-04 by kueblc

iqmend, 2020-02-04:
If i do like this i have to downgrade because i want use 7.1 
And i think downgrade to 7.1 is not possible afer i install 8.1

kueblc, 2020-02-04:
Then why not start with minimal? The limit is not ours, it is a technical limitation that applies to every ESP OTA upgrade.

kueblc, 2020-02-04:
You can also add (`A55AA55A FFFFFFFF 5AA55AA5`) to the end of any firmware to allow Tasmota 8.1 to upgrade to it.

meingraham, 2020-02-04:
Danger Will Robertson! Danger!

Do not flash with tasmota-minimal. You have to start with something that will let you configure the Wi-Fi parameters and minimal will not. This version expects to find the Tasmota settings already in place.

iqmend, 2020-02-04:
absolut right @meingraham 
i tryed one plug with minimal and got an error at wifi conect. Now i can not access to this plug. 
Its a TP22 Plug i can also not open it to flash with ftdi -.-

kueblc, 2020-02-04:
Thanks for chiming in @meingraham, I'll keep that in mind