ct-Open-Source/tuya-convert #531: Bulb will not convert

blackax1, 2020-02-06:
**Did you have issues getting a device to flash?**
Merkury MI-BW210-999W

This is the first device that I could not get to convert.

I have not connected to the smart life app to get what firmware is currently installed.

Im not sure if this might be related to https://github.com/ct-Open-Source/tuya-convert/issues/472

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4163078/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4163079/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4163080/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4163081/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4163082/smarthack-wifi.log)

[capture.pcap.txt](https://github.com/ct-Open-Source/tuya-convert/files/4163085/capture.pcap.txt)

kueblc, 2020-02-06:
I only see one station connect in the wifi log, so it probably didn't connect. Try repeating the pairing process.

blackax1, 2020-02-06:
Did it with two differnt bubls (from the same package) and reset them EZ connect, Im gonna try the dev branch to see maybe something changed.

On one bulb I conencted with the smartlife app but killed app and the power onces it went solid, so I doubt it had enought time to upload a new firmware (plus the issue was going one before the smartlife app and with the bulb that was never connected to the app)

blackax1, 2020-02-06:
Dev branch worked!

closed 2020-02-06 by blackax1

blackax1, 2020-02-06:
Most likey not needed but here are the logs and the firmware
[device-info.txt](https://github.com/ct-Open-Source/tuya-convert/files/4163538/device-info.txt)
[firmware-c1df1c.bin.txt](https://github.com/ct-Open-Source/tuya-convert/files/4163539/firmware-c1df1c.bin.txt)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4163540/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4163541/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4163542/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4163543/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4163544/smarthack-wifi.log)

titilambert, 2020-02-10:
Quick question, when did you buy the bulb ? I tried to convert one with the date: 1907 (it didn't work (stuck at can not push intermediate firmware), and on other one with the date 1807 (it did work)
Thanks for your help !

blackax1, 2020-02-14:
All of mine are 1908