ct-Open-Source/tuya-convert #532: Changing wlan device in docker not working

hadmut, 2020-02-06:
Was just trying to run the docker version as described in the README under Ubuntu 18.04. 

On my machine, the WLAN device is not named wlan, but wlp2s0, so I've changed this in the docker-compose.yml to 
      - WLAN=wlp2s0

but it still issues an error message

Checking for network interface wlan0... Not found!
Please edit WLAN in config.txt to one of: br-2ced10523228, br-96031cad200f, br-a98042af9c91, br-b4b997fb7eaf, docker0, enp4s0, lo, veth07c61aa, virbr0, virbr0-nic, vlan0, wlp2s0
======================================================

Passing the variable through to the config does not work.

kueblc, 2020-02-06:
@Cyber1000

Cyber1000, 2020-02-07:
I'll look into this in the next few hours.

@hadmut: Just 2 questions: 
- Which version? The latest master from here https://github.com/ct-Open-Source/tuya-convert?
- Which architecture of your host? x86, arm

hadmut, 2020-02-08:
Yeah, latest checkout, x86. 

I had more problems, just run the docker file without -d to see the console output, some error message, which might be caused by the underlying phusion/baseimg (some errors with syslogng), and I could not login with the additional wifi client and the device fast enough due to too short script timeouts. It just terminates before I'm able to login with a mobile phone and have the tuya device up.  

However, this two step thing with docker-compose up -d and additional  docker-compose exec tuya start seems somewhat broken, not exactly the docker style and clean design. Usually those things are put in an entrypoint script that configures all those variables at (before) startup time with default values and using the environment variables, to have all things cleanly start and get running at startup time. 

Has this ever been running smoothly?

hadmut, 2020-02-08:
(btw., working well on raspberry.)

Cyber1000, 2020-02-08:
@hadmut 
In short - in DOCKERFILE try this and rebuild your container: 
FROM phusion/baseimage:0.11

I know this has worked before (on ubuntu 18.04)

Long info will follow ...

Cyber1000, 2020-02-09:
Ok just came to test this with the latest release on my ubuntu 18.04 x86 and everything was just working fine, so the `FROM phusion/baseimage:master` shouldn't be the problem.

Now to the points:

> just run the docker file without -d to see the console output

This may only configure things like "network interface" and not start an AP (so you can't login even if you are fast), since we are running the same tuya-script from inside and outside of docker (with all the confirmations it takes)

> However, this two step thing with docker-compose up -d and additional docker-compose exec tuya start seems somewhat broken, not exactly the docker style and clean design.

It's more or less always a 2 step thing with docker: 
- Most time you start a container with docker-compose up -d
- Then you interact with it. Most time you connect via browser, entering https://{dockerhost}:8080 or this like. In this case we already had a working command line script, I didn't rewrite it. So the easiest way was to interact via commandline which means in docker-terms to use an exec command, which is a valid option for docker. An easy webinterface (with start/stop and messages in between) would be my favorite too, but this would involve deeper changes.

> Usually those things are put in an entrypoint script

The last line in the DOCKERFILE (the line with /etc/service/tuya) creates the infrastructure which is executed by the entrypoint script of phusion/baseimage, that's the proposed way phusion/baseimage wants this. So actually `all those variables` are configured in an entrypoint-script through the infrastructure of .phusion/baseimage.

> Has this ever been running smoothly?
Yes all the time

So there must be something different on your setup, I'm starting a new comment for this.

Cyber1000, 2020-02-09:
@hadmut 
To investigate this further:

Number 1:
- docker-compose up -d
- docker-compose logs tuya
- Post the output
-------------------------------------
Number 2:
- Post your docker-compose.yml
--------------------------------------
Number 3:
- did you build the image (docker build -t tuya:latest .) on the same computer or at least same architecture than were you are starting your container (docker-compose up -d)
--------------------------------------
Number 4:
> I had more problems

Could you elaborate on this?

--------------------------------------
That were the mainpoints which came into my mind for now

hadmut, 2020-02-09:
Number 1:

% docker-compose logs tuya
Attaching to docker_tuya_1
tuya_1  | *** Running /etc/my_init.d/00_regen_ssh_host_keys.sh...
tuya_1  | *** Running /etc/my_init.d/10_syslog-ng.init...
tuya_1  | Error parsing source, source plugin unix-dgram not found in /etc/syslog-ng/syslog-ng.conf at line 20, column 8:
tuya_1  | 
tuya_1  |        unix-dgram("/dev/log");
tuya_1  |        ^^^^^^^^^^
tuya_1  | 
tuya_1  | syslog-ng documentation: https://www.balabit.com/support/documentation?product=syslog-ng-ose
tuya_1  | contact: https://lists.balabit.hu/mailman/listinfo/syslog-ng
tuya_1  | *** /etc/my_init.d/10_syslog-ng.init failed with status 1
tuya_1  | 
tuya_1  | *** Killing all processes...
tuya_1  | *** Running /etc/my_init.d/00_regen_ssh_host_keys.sh...
tuya_1  | *** Running /etc/my_init.d/10_syslog-ng.init...
tuya_1  | Error parsing source, source plugin unix-dgram not found in /etc/syslog-ng/syslog-ng.conf at line 20, column 8:
tuya_1  | 
tuya_1  |        unix-dgram("/dev/log");
tuya_1  |        ^^^^^^^^^^
tuya_1  | 
tuya_1  | syslog-ng documentation: https://www.balabit.com/support/documentation?product=syslog-ng-ose
tuya_1  | contact: https://lists.balabit.hu/mailman/listinfo/syslog-ng

That's one of the „more problems” I had, just saw them in docker-compose up without -d
BTW: The phusion/baseimage docker downloaded and used to build is
phusion/baseimage          master              4fc88116db8c        8 months ago        175MB

There is no later one available.

hadmut, 2020-02-09:
Number 2:

% cat docker-compose.yml
version: '3'
services:
  tuya:
    image: tuya:latest
    privileged: true
    network_mode: "host"
    environment:
      - WLAN=wlp2s0
      - AP=vtrust-flash
      - GATEWAY=10.42.42.1
    volumes:
      - ./data/backups:/usr/bin/tuya-convert/backups

Just changed the WLAN name.

hadmut, 2020-02-09:
Number 3: Yes, very same machine. One step after another. 

(Wouldn't it be convenient to use a build path in docker-compose.yml to put building into docker-compose up?)

hadmut, 2020-02-09:
Number 4: I had two more problems. 

One is the syslog-ng problem shown above („Number 1”). 

The second is a timeout in the interaction. 

When doing the start command, tuya-convert asks for user interaction twice. One is to just ask you whether you haven understood ... and enter 'yes'. 

The second asks you to first put some device into the WLAN  network offered by tuya-convert (for enabling that broadcast thing afaik), then put the tuya into blinking configuration mode, and then press enter. 

Both interactions had a timeout after some seconds under docker. On my first try I even took too long to enter 'yes', because wasn't aware that this is time critical. On my second try I entered 'yes' in time, but then was not able to put my android tablet into the WLAN _and_ have the tuya device blinking within the timeout. After timeout the program just terminates. 

When running it on a raspberry, I did not see any timeouts.

Cyber1000, 2020-02-11:
Thanks for your help!

> phusion/baseimage master 4fc88116db8c 8 months ago 175MB

Having exactly the same version, tried it on ubuntu 18.04 and fedora 31 both x86, didn't get the errors. Seems to have problems with the initialization of the pusionimage-part

> docker-compose.yml

Looks fine too

> Wouldn't it be convenient to use a build path in docker-compose.yml to put building into docker-compose up?

I normally separate this, cause bulding docker-containers is not neccessarily on the same machine for me than running them. But I think it would be easier for most users to have only one docker-compose.yml, so I think I'll go for it.

So it's kind of weird, the problems with unix-dgram and the timeouts, that's not a big container and it's not really cpu-consuming, so the timeouts shouldn't happen.

On your host do you have /dev/log?
On my ubuntu I'm getting this with an `ls -lisa /dev/log`
`436 0 lrwxrwxrwx 1 root root 28 Jän 20 18:24 /dev/log -> /run/systemd/journal/dev-log`

ptempier, 2020-02-14:
had the same pb in docker, worked around it via :

outside docker (on the host):
sudo ifconfig   wlxb4750e1f836f down
sudo ip link set wlxb4750e1f836f name wlan0
sudo ifconfig   wlan0 up

Cyber1000, 2020-02-14:
Thanks for your input. After searching the net I think it's a kind of permission problem. I found a similar bug (with unix-dgram("/dev/log") ) on another docker-container with host Linux Solus:  https://forum.seafile.com/t/problems-at-startup/8879
I'll continue with issue #540, cause I assume there's the main problem ...

iregev, 2020-11-05:
Hi
I have RCA for the problem:
config.txt should have the exact wifi name as the docker compose yml has.
If I follow the readme file for the docker setup and prior build command also changes config.txt and then continue the build
everything works fine.

I guess additional changes should be done for enabling non wlan0 for docker

Cyber1000, 2021-01-30:
@iregev If this change to config.txt is necessary there is something wrong with building the image. But I've changed quite a bit (PR is coming soon), hoping that the problem vanishes ...