ct-Open-Source/tuya-convert #534: NeoCoolcam plug died - Device did not appear with the intermediate firmware

odechr, 2020-02-08:
Hi all,

I was trying to flash a NeoCool Cam 16A wifi plug. They should be okey to flash. It is an ESP board.
Mine just died. Before i start flashing a new one.
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4174142/smarthack-wifi.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4174143/smarthack-web.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4174144/smarthack-udp.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4174145/smarthack-psk.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4174146/smarthack-mqtt.log)

This is the message i got:
```
======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
...........................................................................................
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] n
```

And one of the logfiles gives me this: (I just flashed several devices the last days and all worked fine)
```
^CTraceback (most recent call last):

  File "./psk-frontend.py", line 113, in <module>
    main()
  File "./psk-frontend.py", line 106, in main
    r,_,_ =  select.select(readables, [], [])
KeyboardInterrupt

```

kueblc, 2020-02-08:
Don't panic, the device is not dead! It is running the intermediate firmware

You can restart the process, but you do not need to connect a phone or put the device in pairing mode this time. Just press enter and plug the device back in.

> ```
> ^CTraceback (most recent call last):
> 
>   File "./psk-frontend.py", line 113, in <module>
>     main()
>   File "./psk-frontend.py", line 106, in main
>     r,_,_ =  select.select(readables, [], [])
> KeyboardInterrupt
> ```

This is a `KeyboardInterrupt`, ie. you stopped the script, not an error.

odechr, 2020-02-08:
You just saved my day! And im a little embarrassed :) 
I had to keep my phone connected actually or else it didnt wanna flash.

```
Flashed http://10.42.42.1/files/power_plug1.bin successfully in 22009ms, rebooting...

HAVE FUN!
```

closed 2020-02-08 by odechr