ct-Open-Source/tuya-convert #535: Doesn't work on my Raspberry Pi 3 Model B+ with Stretch

depen84, 2020-02-08:
Hi,

I'm writing this feeling a bit lost. I have bought some smart plugs (Blitzwolf SHP2, SHP5, SHP6 15A and SS1) that I would like to convert, but failed. I´ve used tuya-convert v2.3.0 before on SHP6 10A and I remeber that I had some struggle, but at least I made it. Now when I'm doing it again, it doesn't work on my pi3b+ with a clean Stretch installation.

I have tested the 2.4.3 release using: `git clone https://github.com/ct-Open-Source/tuya-convert`
and dev using `git clone --branch development https://github.com/ct-Open-Source/tuya-convert`

I have made multiple restarts and runned the `./install_prereq.sh` multiple times (eg. before restart, after restart but before start flash, after first start flash try, after second start flash etc). 

I have tried 3 different wifi devices:
- Android 9 phone - reconnect's in a loop
- Windows 10 laptop - connects with a pretty solid connection. But, and I believe it's a big but, it can rarely ping 10.42.42.1 and when it happens it only work a few times (like 3-10) before it gets unreachable. I have gotten the webpage connected to vtrust once, and that time it started to backup a SHP2*.
- Tuya-Convert-Donor by digiblur - connects most of the times without any issues. Rock solid connection.

*) That time I had forgotten to copy my esphome bins and I choose to quit in the menu. That device won't ligtup it's led now regardless of power cycling or button clicks. I guess it's bricked..?

I have only added these packages: `sudo apt-get install git usbmount ntfs-3g` and I'm using SSH over ethernet.
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4175353/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4175354/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4175355/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4175356/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4175357/smarthack-wifi.log)

kueblc, 2020-02-08:
> *) That time I had forgotten to copy my esphome bins and I choose to quit in the menu. That device won't ligtup it's led now regardless of power cycling or button clicks. I guess it's bricked..?

Don't panic, it's not bricked, just running the intermediate firmware.

You can restart the process, but you do not need to connect a phone or put the device in pairing mode this time. Just press enter and plug the device back in.

depen84, 2020-02-09:
Thanks, I'll try again.

It bothers me that ping only works for a short time. When I'm on the landingpage connecting my laptop. Ping should atleast work until I click enter and smarthack starts. Or?

kueblc, 2020-02-20:
Hi @depen84 have you been able to recover your device?

> It bothers me that ping only works for a short time. When I'm on the landingpage connecting my laptop. Ping should atleast work until I click enter and smarthack starts. Or?

Sounds like you should use your tuya-convert-donor if your connection has been unreliable otherwise, or see if you can debug `hostapd` on your RPi.

depen84, 2020-04-07:
Never got it working. Soldered my devices instead. Thanks any way!

closed 2020-04-07 by depen84