ct-Open-Source/tuya-convert #536: Vtrust-Recovery backup not downloaded can not reach device

riker65, 2020-02-09:
I got stuck with the convert having wlan vtrust-recovery launched.

it is **a gosund SP1**

A restart of - did a couple of those - of start-flash.sh did not solve it.

```
- message. could not fetch a complete backup - 
continue anyway.
```

after doing so and trying to flash tasmota.

process tries to get info from device-IOT.---take a while saying ready to flash.
after selecting tasmota and y.

----but could not reach it.

can ping 10.42.42.42

vtrust-recovery is still up.

How to solve this 

[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4176918/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4176919/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4176920/smarthack-web.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4176921/smarthack-mqtt.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4176922/smarthack-wifi.log)

Dalli75, 2020-02-10:
Unfortunetely I can not help you, but I can confirm I have exectly the same problem with the Gosund E27 Smart LED Bulb.

I stuck at the same place. Bulb is in wlan vtrust mode and the bulb is pingable. But the fetching the backup and uploading the image fails duo to ther error: "could not reach the device"

riker65, 2020-02-10:
hi please try

this worked for me 
some hints from: [https://github.com/ct-Open-Source/tuya-convert/issues/362](url)

connect to vtrust-recovery
with a different device 
`curl http://10.42.42.42/undo`

delete firmwarebackup in backups folder 

start-flash again....

worked

closed 2020-02-10 by riker65

jayshomebrew, 2021-11-07:
@riker65 Thanks so much, this fixed a bad backup for GDTech model MPV2RO-US (plug).

Pa1nstar, 2023-02-12:
@riker65 thank you so much! finally this was the solution to my problem! :-)