ct-Open-Source/tuya-convert #537: Tried to upgrade Tasmota after the convert now it's not working at all

kpmkpm, 2020-02-10:
Hello all. 

Thank you for this awesome tool.

I have a tuya smart socket with power consumption sensors (Smart Life EU Plug WiFi Control smart Socket Power Metering Timing switch 16A 3680W Voice Control work with Alexa Google IFTT). I managed to flash it to tasmota with tuya-convert. Everything worked fine this far. I tried to upgrade for the latest firmware using tasmota menus firmware upgrade. However, it seemd to do something but nothing happened. I awaited for about ten minutes and (perhaps made an error) took the socket off from the mains and maybe the OTA upgrade was still going on.

Anyway, now as I plug it back on mains, nothing is happening. I tried to put it back to fast flashing state (5 second pressing the button) but nothing happened. I also tried to press it for over 40 seconds and also 1, 2, 3, 4, 5, 6, 7 quick pushes but nothing is happening. Is this socket a brick now, or is there a way to make it work again?

Thanks in advance.

meingraham, 2020-02-10:
Try https://tasmota.github.io/docs/#/help/Device-Recovery?id=fast-power-cycle-device-recovery

kueblc, 2020-02-10:
>  I tried to put it back to fast flashing state (5 second pressing the button) but nothing happened

That is a feature of the stock firmware, so it wouldn't work once you have Tasmota installed.

Thanks @meingraham for your input, hopefully it's helpful for @kpmkpm 

Closing as not a tuya-convert question or issue.

closed 2020-02-10 by kueblc