ct-Open-Source/tuya-convert #538: AP mode not supported! - when it is :/

ozzi91, 2020-02-12:
Hello guys!

first of all i would like to say hello to all of you.

Second, i cannot run  ./start_flash.sh succesfully

This is my log`./start_flash.sh
tuya-convert v2.4.3
./setup_checks.sh: line 32: iw: command not found
AP mode not supported!
Please attach a WiFi card that supports AP mode.
======================================================
Cleaning up...
No screen session found.
No screen session found.
No screen session found.
No screen session found.
Closing AP
Exiting...
`

This is my iw list
`kali@kali:~/tuya-convert$ sudo iw list
Wiphy phy0
        max # scan SSIDs: 20
        max scan IEs length: 195 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports RSN-IBSS.
        Supported Ciphers:
                * WEP40 (00-0f-ac:1)
                * WEP104 (00-0f-ac:5)
                * TKIP (00-0f-ac:2)
                * CCMP-128 (00-0f-ac:4)
                * CCMP-256 (00-0f-ac:10)
                * GCMP-128 (00-0f-ac:8)
                * GCMP-256 (00-0f-ac:9)
        Available Antennas: TX 0 RX 0
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
`

sudo if config -a

`kali@kali:~/tuya-convert$ sudo ifconfig -a
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.10  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::aac3:44e3:5dc:f143  prefixlen 64  scopeid 0x20<link>
        ether 08:2e:5f:7d:1a:cd  txqueuelen 1000  (Ethernet)
        RX packets 76481  bytes 96181859 (91.7 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 43976  bytes 4013486 (3.8 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 20  bytes 864 (864.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 20  bytes 864 (864.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 56:bf:0b:5e:62:2f  txqueuelen 1000  (Ethernet)
        RX packets 49  bytes 8861 (8.6 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 26  bytes 3553 (3.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
`
`kali@kali:~/tuya-convert$ sudo ifconfig -a
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.10  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::aac3:44e3:5dc:f143  prefixlen 64  scopeid 0x20<link>
        ether 08:2e:5f:7d:1a:cd  txqueuelen 1000  (Ethernet)
        RX packets 76481  bytes 96181859 (91.7 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 43976  bytes 4013486 (3.8 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 20  bytes 864 (864.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 20  bytes 864 (864.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 56:bf:0b:5e:62:2f  txqueuelen 1000  (Ethernet)
        RX packets 49  bytes 8861 (8.6 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 26  bytes 3553 (3.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
`
on config i changed name to wlan0. If u need more info please naswer me, i will try to provide it if my knowledge about linux will be enough. I tried fix it like 2 days, but i didnt see any solution

kueblc, 2020-02-12:
This is a curious error:
```
iw: command not found
```
It would explain why it mistakenly determined that AP mode was not supported, because it could not check. Why it was not found is the real problem.

kueblc, 2020-02-12:
PS. Use triple backtick (` ``` `) to enclose multi line code blocks

mydevbox, 2020-02-12:
iw may need to run as root. Try this:

`sudo ./start_flash.sh`

kueblc, 2020-02-12:
Neither `iw list` nor `start_flash` should need root. In any case the error was `not found` rather than a permissions error, baffling as it was clearly run successfully by the user. Unless @ozzi91 installed `iw` after attempting `start_flash`.

mydevbox, 2020-02-12:
right, but the user showed that they ran iw with sudo. I assume that was for a reason. 

`kali@kali:~/tuya-convert$ sudo iw list`

ozzi91, 2020-02-12:
Guys, running sudo ./start_flash.sh worked. Sorry for bothering you. I tried everything but i didnt try the obvious thing. Ty guys!

kueblc, 2020-02-12:
Great. Still curious why there should be any difference. In any case glad to hear you have resolved it.

closed 2020-02-12 by kueblc

DanielO, 2020-03-08:
I found I had this error because /sbin wasn't in the path so I ran it like so:
env PATH=/sbin:$PATH ./start_flash.sh

mhansen, 2020-06-24:
FYI the cause is that `/sbin/iw` is only on the `$PATH` for sudo in new Raspbian distros.

theBasher91, 2020-07-05:
Also seeing something similar:

tuya-convert v2.4.4
AP mode not supported!
Please attach a WiFi card that supports AP mode.
======================================================
Cleaning up...
No screen session found.
No screen session found.
No screen session found.
No screen session found.
Closing AP
Exiting...

Tried sudo, redoing the prereqs script (finished successfully). Any ideas?