ct-Open-Source/tuya-convert #539: LSC Led strip dead after flashing

rolandbreedveld, 2020-02-14:
Tried to flash the LSC Led strip, looks good, but after flashing the device is dead,
somebody any idea how to fix this ?

IoT-device is online with ip 10.42.42.42
Fetching firmware backup
% Total % Received % Xferd Average Speed Time Time Time Current
Dload Upload Total Spent Left Speed
100 1024k 100 1024k 0 0 66202 0 0:00:15 0:00:15 --:--:-- 32235
curl: Saved to filename 'firmware-c43baf.bin'

Getting Info from IoT-device
VTRUST-FLASH 1.5
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: c43baf
MAC: A4:CF:12:C4:3B:AF
BootVersion: 7
BootMode: normal
FlashMode: 1M QIO @ 40MHz
FlashChipId: 1540c8
FlashChipRealSize: 2048K
Active Userspace: user2 0x81000

Ready to flash third party firmware!

For your convenience, the following firmware images are already included in this repository:
Tasmota v8.1.0.2 (wifiman)
ESPurna 1.13.5 (base)

You can also provide your own image by placing it in the /files directory
Please ensure the firmware fits the device and includes the bootloader
MAXIMUM SIZE IS 512KB

Available options:
0) return to stock

flash espurna.bin
flash tasmota.bin
q) quit; do nothing
Please select 0-2: 2
Are you sure you want to flash tasmota.bin? This is the point of no return [y/N] y
Attempting to flash tasmota.bin, this may take a few seconds...
Flashed http://10.42.42.1/files/tasmota.bin successfully in 8880ms, rebooting...
Look for a tasmota-xxxx SSID to which you can connect and configure
Be sure to configure your device for proper function!
HAVE FUN!

kueblc, 2020-02-14:
Unplug the device, plug it back in, and look for the `tasmota-xxxx` SSID.

rolandbreedveld, 2020-02-16:
@kueblc  after 2 reboots,it the SSID finaly pop's up, thanxzz

closed 2020-02-16 by rolandbreedveld