ct-Open-Source/tuya-convert #540: docker libsyslog-ng-3.13.so.0: cannot open shared object file: Permission denied

ptempier, 2020-02-14:
was looking at the logs for another issue , found this :
/tmp/build$ docker-compose logs tuya
Attaching to build_tuya_1
tuya_1  | *** Running /etc/my_init.d/00_regen_ssh_host_keys.sh...
tuya_1  | *** Running /etc/my_init.d/10_syslog-ng.init...
tuya_1  | /usr/sbin/syslog-ng: error while loading shared libraries: libsyslog-ng-3.13.so.0: cannot open shared object file: Permission denied
tuya_1  | *** /etc/my_init.d/10_syslog-ng.init failed with status 1
tuya_1  | 
tuya_1  | *** Killing all processes.

kueblc, 2020-02-14:
@Cyber1000 any ideas?

Cyber1000, 2020-02-14:
Ok I assume that #532 (and this issue) is caused by a permission problem on the host. I had no problem on Ubuntu 18.04.3 LTS (though @hadmut had a problem on his ubuntu 18.04) and on Fedora 31. Raspberry (Rasbian?) seemed to work too with @hadmut.

@ptempier what's your host?

Cyber1000, 2020-02-14:
@ptempier and maybe you have something like SELinux installed on your host, though I have it too on my Fedora (cause it's the default security system on Fedora) and it doesn't have this problem.

Cyber1000, 2020-02-14:
In ubuntu  apparmor is default and I'm having this file to exclude syslog from apparmor: /etc/apparmor.d/disable/usr.sbin.rsyslogd
This file came directly from installation of ubuntu 18.04

ptempier, 2020-02-16:
i run ubuntu and the logs was from a docker guest
i doubt apparmor from host can influence the guest
i ended up running tuya convert on the host
I ordered 4 more sh6 (should receive them in 2-3 weeks), next time i try , i ll look at the apparmor logs

Cyber1000, 2020-03-22:
Any news on this?
And yes, since we run the guest in privileged mode (we won't have access to important host ressources like wifi) we do have full access to the host, so things like apparmor would have influence from host to guest.
You may look if you have /etc/apparmor.d/disable/usr.sbin.rsyslogd on your host.
Sorry for the late reply, last weeks were somehow chaotic ...

sknebel, 2020-03-23:
As a data point, I had a potentially similar issue with access to the syslog-ng control socket that was resolved by temporarily disabling AppArmor (my host was openSUSE)

ptempier, 2020-03-23:
Yes, i have  /etc/apparmor.d/disable/usr.sbin.rsyslogd

I ll close that issue because i don't think i ll be able to look at it closely during the next weeks.
Also most of the required service ports are already used on the host, so docker or not, it's bothersome.

Anyway the issue was mostly "cosmetic", i'll close it, so you can concentrate on more important stuff.
If it cause any real problem or if i have the actual time to look it, i'll re-open it.

closed 2020-03-23 by ptempier