ct-Open-Source/tuya-convert #542: Device does not appear with the intermediate firmware.

GijsTimmers, 2020-02-16:
Running commit 31856ada24a1e951b4da849bff2ed056f544c51b on an Ubuntu 19.10 live usb.
Trying to flash a LSC movement sensor (https://www.mikrocontroller.net/attachment/426891/Bauteilseite1_klein.jpg)

It does not seem to flash the firmware. It starts the web server fine, I can connect to it, SmartConfig completes and then the device does not appear with the intermediate firmware.

All help is appreciated.

Logs:

```
root@ubuntu:/home/ubuntu/tuya-convert/scripts# cat smarthack-mqtt.log 
1581864064: mosquitto version 1.6.6 starting
1581864064: Using default config.
1581864064: Opening ipv4 listen socket on port 1883.
1581864064: Opening ipv6 listen socket on port 1883.
1581864101: New connection from 10.42.42.1 on port 1883.
1581864101: New client connected from 10.42.42.1 as 038880662462ab111c2b (p1, c1, k60, u'038880662462ab111c2b').
1581864101: No will message specified.
1581864101: Sending CONNACK to 038880662462ab111c2b (0, 0)
1581864101: Received SUBSCRIBE from 038880662462ab111c2b
1581864101:     smart/device/in/038880662462ab111c2b (QoS 0)
1581864101: 038880662462ab111c2b 0 smart/device/in/038880662462ab111c2b
1581864101: Sending SUBACK to 038880662462ab111c2b
1581864111: New connection from 127.0.0.1 on port 1883.
1581864111: New client connected from 127.0.0.1 as auto-55783521-FDFF-89B1-1C7B-072AF80ABF1B (p2, c1, k60).
1581864111: No will message specified.
1581864111: Sending CONNACK to auto-55783521-FDFF-89B1-1C7B-072AF80ABF1B (0, 0)
1581864111: Received PUBLISH from auto-55783521-FDFF-89B1-1C7B-072AF80ABF1B (d0, q0, r0, m0, 'smart/device/in/038880662462ab111c2b', ... (111 bytes))
1581864111: Sending PUBLISH to 038880662462ab111c2b (d0, q0, r0, m0, 'smart/device/in/038880662462ab111c2b', ... (111 bytes))
1581864111: Received DISCONNECT from auto-55783521-FDFF-89B1-1C7B-072AF80ABF1B
1581864111: Client auto-55783521-FDFF-89B1-1C7B-072AF80ABF1B disconnected.
1581864192: Client 038880662462ab111c2b has exceeded timeout, disconnecting.
1581864225: New connection from 10.42.42.1 on port 1883.
1581864225: New client connected from 10.42.42.1 as 038880662462ab111c2b (p1, c1, k60, u'038880662462ab111c2b').
1581864225: No will message specified.
1581864225: Sending CONNACK to 038880662462ab111c2b (0, 0)
1581864225: Received SUBSCRIBE from 038880662462ab111c2b
1581864225:     smart/device/in/038880662462ab111c2b (QoS 0)
1581864225: 038880662462ab111c2b 0 smart/device/in/038880662462ab111c2b
1581864225: Sending SUBACK to 038880662462ab111c2b
1581864235: New connection from 127.0.0.1 on port 1883.
1581864235: New client connected from 127.0.0.1 as auto-9C5F5CEE-B1FE-9421-D441-EA9EF89A2892 (p2, c1, k60).
1581864235: No will message specified.
1581864235: Sending CONNACK to auto-9C5F5CEE-B1FE-9421-D441-EA9EF89A2892 (0, 0)
1581864235: Received PUBLISH from auto-9C5F5CEE-B1FE-9421-D441-EA9EF89A2892 (d0, q0, r0, m0, 'smart/device/in/038880662462ab111c2b', ... (111 bytes))
1581864235: Sending PUBLISH to 038880662462ab111c2b (d0, q0, r0, m0, 'smart/device/in/038880662462ab111c2b', ... (111 bytes))
1581864235: Received DISCONNECT from auto-9C5F5CEE-B1FE-9421-D441-EA9EF89A2892
1581864235: Client auto-9C5F5CEE-B1FE-9421-D441-EA9EF89A2892 disconnected.
1581864316: Client 038880662462ab111c2b has exceeded timeout, disconnecting.
1581864318: New connection from 10.42.42.1 on port 1883.
1581864318: New client connected from 10.42.42.1 as 038880662462ab111c2b (p1, c1, k60, u'038880662462ab111c2b').
1581864318: No will message specified.
1581864318: Sending CONNACK to 038880662462ab111c2b (0, 0)
1581864318: Received SUBSCRIBE from 038880662462ab111c2b
1581864318:     smart/device/in/038880662462ab111c2b (QoS 0)
1581864318: 038880662462ab111c2b 0 smart/device/in/038880662462ab111c2b
1581864318: Sending SUBACK to 038880662462ab111c2b
^C1581864320: Socket error on client 038880662462ab111c2b, disconnecting.
1581864320: mosquitto version 1.6.6 terminating
```

```
root@ubuntu:/home/ubuntu/tuya-convert/scripts# cat smarthack-psk.log 
new client on port 443 from 10.42.42.39:46960
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1076)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.39:46962
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1076)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.39:46964
[... 186 lines cut ...]
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlo1...
Configuration file: /dev/stdin
Using interface wlo1 with hwaddr 98:2c:bc:ab:54:5a and ssid "vtrust-flash"
wlo1: interface state UNINITIALIZED->ENABLED
wlo1: AP-ENABLED 
wlo1: AP-STA-CONNECTED 72:f0:0c:e2:6e:2a
wlo1: AP-STA-CONNECTED 24:62:ab:11:1c:2b
wlo1: AP-STA-DISCONNECTED 24:62:ab:11:1c:2b
wlo1: AP-STA-CONNECTED 24:62:ab:11:1c:2b
wlo1: AP-STA-DISCONNECTED 24:62:ab:11:1c:2b
wlo1: AP-STA-CONNECTED 24:62:ab:11:1c:2b
wlo1: AP-STA-DISCONNECTED 24:62:ab:11:1c:2b
wlo1: AP-STA-CONNECTED 24:62:ab:11:1c:2b
wlo1: interface state ENABLED->DISABLED
wlo1: AP-STA-DISCONNECTED 24:62:ab:11:1c:2b
wlo1: AP-STA-DISCONNECTED 72:f0:0c:e2:6e:2a
wlo1: AP-DISABLED 
wlo1: CTRL-EVENT-TERMINATING 
nl80211: deinit ifname=wlo1 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
Restarting NetworkManager...
```

The-May, 2020-02-16:
Same issue here with a pi3 on the latest raspbian buster version 2020-02-05 and 2019-04-08 raspbian stretch.

kueblc, 2020-02-16:
Duplicate of #135. These devices have a secondary MCU that powers down the ESP until there is data to send, so it is a tricky matter to keep the device online long enough to flash.

closed 2020-02-16 by kueblc