ct-Open-Source/tuya-convert #544: New bulbs/lamps, troubleshooting the process

ltlw0lf, 2020-02-16:
This set of scripts is awesome, and I've flashed a number of my tuya bulbs (Novastella and Slitinto.)

I am having problems with some new bulbs I have purchased from Amazon, as well as some older lamps, and it appears this may be a change in the firmware or something isn't quite right in the process.  None of these have been attached to the app, but they were all purchased after the new year.  I am using a Raspbian Stretch on a Raspberry Pi 3b+ (and it has worked about a dozen times.)  All of the bulbs and lamps exhibit the same flaw in that they take an ip address from the dhcp server and do not use the default 10.42.42.42 (even though I have a device connected to the AP and pinging 10.42.42.42.)  I broke apart the script to see what is happening, and it appears that the device connects to the ap successfully, grabs an IP address, and then connects to psk_frontend.py.  I then get the message "a bytes-like object is required, not 'str'."

On the Novastella 20W IP66 (ec:fa:bc:xx:xx:xx), I get the following...

From start_ap, 
wlan0: AP-STA-CONNECTED 34:e1:2d:xx:xx:xx <- Laptop
wlan0: AP-STA-CONNECTED ec:fa:bc:xx:xx:xx <- Device

I then see from tcpdump:

10.42.42.19.68 > 10.42.42.1.67 DHCP, Request from ec:fa:bc:xx:xx:xx
[...]
Hostname Option 12, length 10: "ESP_48C6C631"
[...]
10.42.42.1.67 > 10.42.42.19.68 DHCP, Reply from b8:27:eb:xx:xx:xx

I then see what appears to be encrypted traffic from 10.42.42.19 to 10.42.42.1.

I am seeing the same from brand new Slitinto E26 9W bulbs (which I will attach to logs later.)

[fake-registration-server.log](https://github.com/ct-Open-Source/tuya-convert/files/4210867/fake-registration-server.log)
[psk_frontend.log](https://github.com/ct-Open-Source/tuya-convert/files/4210870/psk_frontend.log)
[setup_ap.log](https://github.com/ct-Open-Source/tuya-convert/files/4210871/setup_ap.log)
[output.dump.log](https://github.com/ct-Open-Source/tuya-convert/files/4210873/output.dump.log)

ltlw0lf, 2020-02-16:
Slitinto TB95 bulbs exhibiting the same issue (purchased from Amazon last week.)

[fake-registration-server.log](https://github.com/ct-Open-Source/tuya-convert/files/4210904/fake-registration-server.log)
[psk_frontend.log](https://github.com/ct-Open-Source/tuya-convert/files/4210905/psk_frontend.log)
[setup_ap.log](https://github.com/ct-Open-Source/tuya-convert/files/4210906/setup_ap.log)
[output.dump.log](https://github.com/ct-Open-Source/tuya-convert/files/4210908/output.dump.log)

iracigt, 2020-02-19:
I actually ran into a similar issue last night on a pair of brand new Slitinto TB95 bulbs attempting to use tuya-convert for the first time. I ran mine from a Raspberry Pi Zero W running a fresh install of Raspbian.  I cracked open a bulb and can confirm there is an ESP8285 inside. The module bears the marking "ajw-05_8285" on the silk screen. There are test points for TX and RX, but they aren't easily accessible. I may need to desolder the ESP8285 module. I can upload my logs later today if that's helpful.

kueblc, 2020-02-19:
@ltlw0lf have you updated tuya-convert or is this a clean install? Sounds like #500
```
a bytes-like object is required, not 'str'
```
This is symptomatic of a mismatch between the installed `sslpsk` and tuya-convert.

Try doing a `git pull` followed by `./install_prereq.sh`.

ltlw0lf, 2020-02-20:
No upgrade, fresh install that converted 11 bulbs purchased in November and December during the same time it failed with four new bulbs (purchased a week and a half ago).  Old bulbs do not exhibit the same issue.  However, I will delete the directory and re-git and reinstall to be sure as it sounds plausible that something might have gone corrupt.

michaelkuty, 2020-02-20:
I have same issue, bough 4 bulbs and only I’m able flash only one, another just keep rapidly blinking and after I start flash it switched blinking to stale light and continously retrying without any success

kueblc, 2020-02-20:
@ltlw0lf 

> Old bulbs do not exhibit the same issue

If they are old they might not be running TLS and thus wouldn't be affected by `psk-frontend` issues. Let me know how it works out after upgrade.

@michaelkuty 

> I think they disabled OTA in new firmaware

It's not new firmware with OTA disabled, we just can't flash a different microcontroller with ESP firmware as this would brick the device.

> WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware

michaelkuty, 2020-02-20:
Hm but these bulbs are from one seller in same batch 1 flashed without any issues and three with this issue, I think that the new release is with new board inside ?

kueblc, 2020-02-20:
> Hm but these bulbs are from one seller in same batch 1 flashed without any issues and three with this issue, I think that the new release is with new board inside ?

Sounds like you just got lucky they had one old bulb in stock, newly manufactured devices are now using other microcontrollers.

ltlw0lf, 2020-02-22:
Re-git of the code resulted in the error going away, but now the bulb keeps asking the DHCP server for an IP address and gets nothing back.  Will try to figure out why dnsmasq isn't assigning IP addresses.

kueblc, 2020-02-22:
> Ok, restarted and everything is working, but bulb is still getting 10.42.42.40, not 10.42.42.42, so the code isn't able to complete the upload of the firmware. I am going to try to modify the code to use 10.42.42.40 instead.

Don't do this. The bulb will request `10.42.42.42` if and only if it has downloaded and run the intermediate firmware. Let's take a look at your most recent logs and see why it is not running the intermediate firmware.

ltlw0lf, 2020-02-22:
Laptop is still getting 10.42.42.25, newest logs attached along with tcpdump output.

[output.dump.log](https://github.com/ct-Open-Source/tuya-convert/files/4239798/output.dump.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4239799/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4239800/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4239801/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4239802/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4239803/smarthack-udp.log)

kueblc, 2020-02-22:
Unfortunately, I have found your problem
```
new client on port 443 from 10.42.42.40:59911
ID: 0242416f68626d64366147393149465231042860c5307ae18dd03e0a23dfed0dc2a3db9b8f0b7a0e7ec58f558fc1571873
PSK: d1b2cb80cb023ca40a924400067b7976c11f6b5dd0f41d5ba026496ddf6c8f5c
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:720)
```
This is the latest patched firmware from Tuya, which uses a yet-uncracked PSK. See #483 

Sorry I wish I had better news for you, it is unfortunately not compatible with tuya-convert at this time.

ltlw0lf, 2020-02-22:
That is quite alright...I will stand by and wait to see if this changes.  I was able to get the IP66 20W ones working with the correct code, just these ones aren't working.

Wish they would stop playing with the firmware and allow us to use whatever firmware we wanted.

closed 2020-02-22 by ltlw0lf