ct-Open-Source/tuya-convert #545: now devices death

maestroat, 2020-02-19:
nice tool nice description

thank you now devices short light blue when connect to electricity and than nowhing

flash works finde all looks good but than death.

### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
_Please add it to the wiki instead!_

**Do you have a question about configuring your newly flashed device?**
_Please ask in your alternative firmware's support channel instead!_

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

**Please remember to close your issue when the problem has been addressed!**

kueblc, 2020-02-19:
Please follow the instructions laid out before you, you need to include your logs to help determine the issue.

kueblc, 2020-02-19:
With no other information it sounds like your device is running the intermediate firmware. You can recover from this by restarting the process, as written in the Troubleshooting wiki.

kueblc, 2020-02-20:
Closing as unclear, unresponsive, and not enough information to debug.

closed 2020-02-20 by kueblc