ct-Open-Source/tuya-convert #547: Flashing lightbulbs - wlan0: Could not connect to kernel driver

dominikheinz, 2020-02-20:
I am trying to flash the Fcmila light bulbs with _tuya-convert_.
Here is a link to the product: https://templates.blakadder.com/fcmila-spotlight-6w.html

The _tuya-convert_ procedure was attempted on a raspberry pi 3B+.
I did the following steps:

1. Start the `start_flash.sh` script.
2. Continue till it prompts to put the bulbs into pairing mode and also to connect to vtrust-flash.
3. The bulbs could be put into pairing mode by turning them on and off a few times (they blink).
4. No vtrust-flash access point is visible, hence the procedure doesn't work when continuing:

```
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
.Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
...............
SmartConfig complete.
Resending SmartConfig Packets
...............
SmartConfig complete.
Resending SmartConfig Packets
...............
SmartConfig complete.
Resending SmartConfig Packets
...............
SmartConfig complete.
Resending SmartConfig Packets
...............
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
...............
SmartConfig complete.
Resending SmartConfig Packets
```

Checking the logs, I am being told, it wasn't possible to connect to the kernel driver of wlan0.
Please see the attached logs.

[logs.zip](https://github.com/ct-Open-Source/tuya-convert/files/4232091/logs.zip)

The main issue seems to be here:
```
Configuring AP interface...
RTNETLINK answers: File exists
RTNETLINK answers: Operation not possible due to RF-kill
Error: Device for nexthop is not up.
Error: Device for nexthop is not up.
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
rfkill: WLAN soft blocked
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:93:cc:29 and ssid "vtrust-flash"
Failed to set beacon parameters
wlan0: Could not connect to kernel driver
Interface initialization failed
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: Unable to setup interface.
wlan0: interface state DISABLED->DISABLED
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface wlan0 wasn't started
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
```

kueblc, 2020-02-20:
> wlan0: Could not connect to kernel driver

Is not an issue

> rfkill: WLAN soft blocked

This is your issue. Do `rfkill unblock wifi`.

dominikheinz, 2020-02-20:
That worked. Thanks !!

closed 2020-02-20 by dominikheinz

dominikheinz, 2020-02-20:
@kueblc I flashed 2 of 3 bulbs successfully.
However, for the last one it seems as I can not properly connect to the vtrust-flash access point. 
When connecting with my phone it connects, and after a second it disconencts again.
The logs confirm that:

```
Configuring AP interface...
RTNETLINK answers: File exists
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:93:cc:29 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: AP-STA-CONNECTED f2:1a:60:ba:73:00
wlan0: AP-STA-DISCONNECTED f2:1a:60:ba:73:00
wlan0: AP-STA-CONNECTED f2:1a:60:ba:73:00
wlan0: AP-STA-DISCONNECTED f2:1a:60:ba:73:00
wlan0: AP-STA-CONNECTED f2:1a:60:ba:73:00
wlan0: AP-STA-DISCONNECTED f2:1a:60:ba:73:00
wlan0: AP-STA-CONNECTED f2:1a:60:ba:73:00
wlan0: AP-STA-DISCONNECTED f2:1a:60:ba:73:00
wlan0: AP-STA-CONNECTED f2:1a:60:ba:73:00
wlan0: AP-STA-DISCONNECTED f2:1a:60:ba:73:00
wlan0: AP-STA-CONNECTED f2:1a:60:ba:73:00
wlan0: AP-STA-DISCONNECTED f2:1a:60:ba:73:00
wlan0: AP-STA-CONNECTED f2:1a:60:ba:73:00
wlan0: AP-STA-DISCONNECTED f2:1a:60:ba:73:00
wlan0: AP-STA-CONNECTED f2:1a:60:ba:73:00
wlan0: AP-STA-DISCONNECTED f2:1a:60:ba:73:00
wlan0: AP-STA-CONNECTED 00:c2:c6:ee:25:7d
wlan0: AP-STA-CONNECTED ec:fa:bc:b6:1b:d4
wlan0: AP-STA-DISCONNECTED 00:c2:c6:ee:25:7d
wlan0: AP-STA-CONNECTED 00:c2:c6:ee:25:7d
wlan0: AP-STA-DISCONNECTED ec:fa:bc:b6:1b:d4
wlan0: AP-STA-CONNECTED ec:fa:bc:b6:1b:d4
wlan0: AP-STA-DISCONNECTED ec:fa:bc:b6:1b:d4
```
Is there something I can do? I can't stay connected long enough to get the flashing procedure started.

kueblc, 2020-02-20:
Is your phone reporting why it disconnected? Some may disconnect when it notices that there is no internet. Glad you got it working with a different device.