ct-Open-Source/tuya-convert #550: Device did not appear with the intermediate firmware with AWP07L

seritos, 2020-02-21:
Hi All, 

I have just succesfully flashed two Maxcio YX-DE02 samrtplugs with the tuya converter. 
However, there is no way of flashing my AWP07L. 
I tried 5 times and always get this error: Device did not appear with the intermediate firrmware.

Any suggestions? 

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4237509/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4237511/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4237512/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4237513/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4237514/smarthack-wifi.log)

thenightfighter, 2020-02-29:
Same here. Checked 4 plugs and all of them have an ESP8266EX inside... it seems that something is wrong with tuya-convert :(

pilotseye, 2020-03-21:
Any new here? It seems like I am having the same issue with tuya-convert and two of my plugs :( is there any way to recover them from the current state? Thanks!

coro96, 2020-04-30:
I was trying to flash a Houzetek AWP07L (old model with no power monitoring) on Ubuntu 20.04 in VirtualBox with a WiFi adapter and I was experiencing the same issue. Then I remembered that the plug is not able to connect to 5 GHz, so I swapped the adapter with an old one that only uses 2.4 GHz and everything worked flawlessly! I hope this can be helpful for you.