ct-Open-Source/tuya-convert #551: Unable to connect to vtrust-flash

jayanthb007, 2020-02-22:
Hey @kueblc!

Thanks for the amazing work you have done on this project and your continued support.

I'm having issues connecting to the SSID vtrust-flash which is being broadcasted, my phone says unable to connect.
I've tried with both **Ubuntu 18.04** and **Raspberry Pi desktop** running on **Oracle VM VirtualBox**. For the wifi adapter I'm using **Ralink RT5370**. I've tried both docker image and regular start_flash.sh method and got stuck at the same step where I'm supposed to connect to vtrust-flash on my phone.

Let me know if you need more information. Appreciate your help and support.

Thanks
Jay

Checking for network interface wlan0... Found.
Checking UDP port 53... Occupied by dnsmasq with PID 2227.
Port 53 is needed to resolve DNS queries
Do you wish to terminate dnsmasq? [y/N] y
Attempting to stop dnsmasq.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Occupied by mosquitto with PID 2484.
Port 1883 is needed to run MQTT
Do you wish to terminate mosquitto? [y/N] y
Attempting to stop mosquitto.service
Checking TCP port 8886... Available.
sudo: ufw: command not found
======================================================
  Starting AP in a screen...
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue
^C======================================================

kueblc, 2020-02-22:
Hi Jay,

Can you check your `/scripts` and `/backup` directories to see if you have any `*.log` files?

jayanthb007, 2020-02-22:
Hi Colin,

Thanks for a quick turnaround on this. I'm attaching those logs.

Jay

jayanthb007, 2020-02-22:
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4240268/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4240269/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4240270/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4240271/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4240272/smarthack-web.log)

Novocaine85, 2020-02-23:
i have the same problem. tried with 2 different devices.

using debian stretch with wifi adaptor.

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4241313/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4241314/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4241315/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4241316/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4241317/smarthack-wifi.log)

kueblc, 2020-02-23:
Are you able to successfully use the hotspot feature of the OS?

kueblc, 2020-02-23:
If you have [`create_ap`](https://github.com/oblique/create_ap) installed, you could also try the `create-ap` fork.
```
git clone -b create-ap https://github.com/kueblc/tuya-convert.git
```

jayanthb007, 2020-02-23:
> Are you able to successfully use the hotspot feature of the OS?

Was that question for me? I haven't used it

jayanthb007, 2020-02-23:
@kueblc: I just bought a raspberry pi 4b and installed Raspbian Buster Lite on it and ran the start_flash but this time I don't even see the vtrust_flash ssid. Please see the log attached. Am I missing wifi drivers?
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4242636/smarthack-wifi.log)

jayanthb007, 2020-02-23:
> @kueblc: I just bought a raspberry pi 4b and installed Raspbian Buster Lite on it and ran the start_flash but this time I don't even see the vtrust_flash ssid. Please see the log attached. Am I [...]

It's fixed and I was able to successfully flash one of the tuya switches I had. Not sure exactly what I did to fix it but couple of reboots and setting up wifi network through raspi-config fixed it for me.

I will attempt to flash it on the VM again just to see if it works and post my update just so it helps others who are having the same issue

jayanthb007, 2020-02-24:
> If you have [`create_ap`](https://github.com/oblique/create_ap) installed, you could also try the `create-ap` fork.
> 
> ```
> git clone -b create-ap https://github.com/kueblc/tuya-convert.git
> ```
Still unable to connect to vtrust-flash. What am I missing?
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4242795/smarthack-wifi.log)

kueblc, 2020-02-24:
I did note this in your log:
```
rfkill: WLAN soft blocked
```
Which you can address with `rfkill unblock wifi`

Baldorian44, 2020-02-24:
Hello, I seem to have the same problem, i've attached my file - although it seems related to "wlan0 could not connect to kernel driver" error instead, any idea how to fix that mr wise kueblc :)

[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4246013/smarthack-wifi.log)

I tried your command with "rfkill unblock wifi" which seems to have removed that block, now i have the issue with the kernel...

kueblc, 2020-02-24:
Don't worry about:
```
wlan0: Could not connect to kernel driver
```
This is not a problem unto itself.

Do you see `vtrust-flash` SSID from another device?

Baldorian44, 2020-02-24:
Ok cool.
No i dont, i've tried to search for it under WiFi settings under two different android phones and 1 laptop.. nothing shows up in the available WiFi's, I have even tried to make a manual network, no luck.

It should just appear as a standard WiFi network right?

Baldorian44, 2020-02-24:
Yeah now i see it, seems like your command worked after a bit of patience :)
Now i just have to figure out the last step of flashing the Gosund plugs :)
Thanks!

jayanthb007, 2020-02-25:
> Still unable to connect to vtrust-flash. What am I missing?
> [smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4242795/smarthack-wifi.log)

@kueblc I noticed that if I remove the setting **--redirect-to-localhost** setting in the line 20 of start_flash.sh. I'm able to connect to vtrust-flash with the VM approach but I'm still unable to flash. Does this help to understand what could be causing the issue?

kueblc, 2020-02-25:
This could mean that the device you are connecting with expects an internet connection, and disconnects when it realizes there is none. However this is intended, we do not want a connection to the outside internet, so you must force the device to stay connected anyway.

With my Android device, it might say something like "No internet connection, won't automatically reconnect", but if I manually reconnect after that it will stay.

Do any other logs show activity?

jayanthb007, 2020-02-25:
> This could mean that the device you are connecting with expects an internet connection, and disconnects when it realizes there is none. However this is intended, we do not want a connection to the [...]

i'm using an iPhone to connect to the AP and it does allow to connect to networks which doesn't have internet connections. I have tried to connect to vtrust-flash from my windows laptop and it says unable to connect too. I know that the usb wifi adapter is capable of hosting an AP. I have tried following create_ap examples and I'm able to connect to the ap i made up.
Bridged Internet sharing:
create_ap -m bridge wlan0 eth0 testnet 
Bridged Internet sharing (pre-configured bridge interface):
create_ap -m bridge wlan0 br0 testnet

 Also if I comment the below line it aloows me to connect to vtrust-flash. My guess is there is something happening in redirecting to the fake webserver created at 10.42.42.1

$screen_with_log smarthack-web.log -S smarthack-web -m -d ./fake-registration-server.py

I will attach the logs in the next reply.

wifi adapter that I am using
https://www.canakit.com/raspberry-pi-wifi.html

jayanthb007, 2020-02-25:
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4251548/smarthack-mqtt.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4251549/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4251550/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4251551/smarthack-wifi.log)

jayanthb007, 2020-02-25:
Windows events:

 Interface name: Intel(R) Dual Band Wireless-AC 8265
 Interface type: Native Wi-Fi
 Profile: Discovery connection
 SSID: vtrust-flash
 SSID length: 12
 Connection mode: Infra
 Security: No
 Connect even if network is not broadcasting: No

Result of diagnosis: Problem found

Root cause:
Wireless association to "vtrust-flash" failed for an unknown reason

Detailed root cause:
802.11 connection failed due to status code 1: unspecified failure

Repair option:
Try connecting to "vtrust-flash" again
If the problem continues, try resetting the access point, or contact the network administrator or your hardware manufacturer for further assistance.

Root Cause: Wireless association to "vtrust-flash" failed for an unknown reason 

Root Cause Guid: {5c92e4e9-4699-418c-9db6-82d57dcbf59f} 

Repair option: Try connecting to "vtrust-flash" again
If the problem continues, try resetting the access point, or contact the network administrator or your hardware manufacturer for further assistance.

amouses, 2020-02-29:
For me the whole procedure is not working and I've debugged it down to the vtrust-flash SSID

I understand that its using hostapd to create the  SSID and dnsmasq to provide the dhcp addresses.

What happens for me is that when I first try to connect my smartphone to the vtrust-flash network I don't get a sustained connection under Android 10.  Andoid 10 keeps trying and failing.   I tired an iOS smartphone, but it too is having difficulty.

I am using a Pi 3B+ with Buster and all fixes applied

---

Presumably if the smartphone (first device) can get a WiFo connection with valid  10.42.42..x address then it is pointless going to the next step where you try to talk to the power switch (in my case)

---   log of session ---

# ./start_flash.sh
tuya-convert v2.4.3
Checking for network interface wlan0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
sudo: ufw: command not found
======================================================
  Starting AP in a screen
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash

NOW I SHOW YOU LOGFILE

# tail -f smarthack-wifi.log
RTNETLINK answers: File exists
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:12:75:b9 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: AP-STA-CONNECTED c0:ee:fb:4a:ab:cf
wlan0: AP-STA-DISCONNECTED c0:ee:fb:4a:ab:cf
wlan0: AP-STA-CONNECTED 20:a2:e4:78:56:ef

---

I notice on the iPhone  the IP address that is shown is not in the correct range, and then on subsequent trying I get 
Unable to join the network "vtrust-flash" on iPhone

---

So back to the point....   Pi 3B+ with buster latest,   ran your script  (that used to work fine), now the sticking point seems that cant first attach with smartphone, and rest of script goes nowhere, i.e. to be never finds by teckin sp23 to flash    (and of course I checked its the oldvariety using tuya-discovery.py )

claell, 2020-03-01:
> I did note this in your log:
> 
> ```
> rfkill: WLAN soft blocked
> ```
> 
> Which you can address with `rfkill unblock wifi`

I ran into the same issue and this fixed it for me! Just wanted to let you know.

dcolling, 2020-03-03:
I've got an RPi4, I seem to have similar issues as above, no vtrust-flash network appearing. And the same start_flash log:
pi@raspberrypi:~/tuya-convert $ ./start_flash.sh
tuya-convert v2.4.3
Checking for network interface wlan0... Found.
Checking UDP port 53... Occupied by dnsmasq with PID 445.
Port 53 is needed to resolve DNS queries
Do you wish to terminate dnsmasq? [y/N] y
Attempting to stop dnsmasq.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Occupied by mosquitto with PID 395.
Port 1883 is needed to run MQTT
Do you wish to terminate mosquitto? [y/N] y
Attempting to stop mosquitto.service
Checking TCP port 8886... Available.
sudo: ufw: command not found

Apologies as I'm really new to this, but I cannot seem to figure out how to obtain the log files.

kueblc, 2020-03-03:
@dcolling check inside your `tuya-convert/scripts/` directory for `.log` files.

computergeek1507, 2020-03-17:
running 'sudo rfkill unblock all' fixed this issue for me

Siytek, 2020-03-25:
I have the same issue on RPi 4. I have flashed many devices with no problem using Ubuntu running on an old MacBook Pro, but I cannot connect to vtrust-flash with the Pi.

Configuring AP interface...
RTNETLINK answers: File exists
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr dc:a6:32:66:c6:df and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: AP-STA-CONNECTED 34:42:62:e3:64:9a
wlan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED 34:42:62:e3:64:9a
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...

Siytek, 2020-03-25:
> I have the same issue on RPi 4. I have flashed many devices with no problem using Ubuntu running on an old MacBook Pro, but I cannot connect to vtrust-flash with the Pi. I can see vtrust-flash but a connection is never established. I have tried with iPhone and also with a script running on another ESP8266.
> 
> Configuring AP interface...
> RTNETLINK answers: File exists
> RTNETLINK answers: File exists
> Starting DNSMASQ server...
> Starting AP on wlan0...
> Configuration file: /dev/stdin
> wlan0: Could not connect to kernel driver
> Using interface wlan0 with hwaddr dc:a6:32:66:c6:df and ssid "vtrust-flash"
> wlan0: interface state UNINITIALIZED->ENABLED
> wlan0: AP-ENABLED
> wlan0: AP-STA-CONNECTED 34:42:62:e3:64:9a
> wlan0: interface state ENABLED->DISABLED
> wlan0: AP-STA-DISCONNECTED 34:42:62:e3:64:9a
> wlan0: AP-DISABLED
> wlan0: CTRL-EVENT-TERMINATING
> nl80211: deinit ifname=wlan0 disabled_11b_rates=0
> AP closed
> Stopping DNSMASQ server...

I managed to get it working, I confirmed hostapd was working ok by doing [this](https://github.com/ct-Open-Source/tuya-convert/issues/373#issuecomment-549988077). It seemed to take a lot longer to connect to vtrust-flash than it does with my Ubuntu box, so I am not sure if I was just being impatient.

therealhammer, 2020-07-04:
I had kind of the same problem. I could (sometimes) see the SSID of the vtrust-flash.

What worked for me (after many hours of not finding the bug whatsoever) was changing the Wifi channel of the network. You can do this by changing the value of "channel" in the file "scripts/hostapd.conf" to a value between 1 and 14 that fits best for your environment/neighborhood. I used the Wifi Analyzer app for finding empty channels. (https://play.google.com/store/apps/details?id=com.farproc.wifi.analyzer&hl=en)

I live in a house with some shitty routers that wont search for empty channels but just use channel 1. So 3 other routers where simply just "louder" than the little wifi dongle I bought. 

Thought i'd share it in case someone runs into the same issue.

jenniferlee1818, 2020-10-22:
I've tried all of the suggestions except for the create-ap.  None of my devices will connect to the vtrust-flash AP.  Running tuya-convert on a fresh Debian Buster with the Ralink RT5370 usb2.0 wifi adapter.

_smarthack-mqtt.log_
1603387165: mosquitto version 1.5.7 starting
1603387165: Using default config.
1603387165: Opening ipv4 listen socket on port 1883.
1603387165: Opening ipv6 listen socket on port 1883.

_smarthack-psk.log_

_smarthack-udp.log_
Listening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667

_smarthack-web.log_
Listening on 10.42.42.1:80

_smarthack-wifi.log_
System info
===========
94acc3b
Linux debian 4.19.0-12-amd64 #1 SMP Debian 4.19.152-1 (2020-10-18) x86_64 GNU/Linux
OpenSSL 1.1.1d  10 Sep 2019
Dnsmasq version 2.80  Copyright (c) 2000-2018 Simon Kelley
Compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile

This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.
hostapd v2.8-devel
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2019, Jouni Malinen <j@w1.fi> and contributors
Python 3.7.3
===========
./setup_ap.sh: line 19: rfkill: command not found
Attempting to stop wpa_supplicant
Stopping NetworkManager...
Configuring AP interface...
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlx70f11c4bf34e...
Configuration file: /dev/stdin
Using interface wlx70f11c4bf34e with hwaddr 72:e0:a8:0c:cc:96 and ssid "vtrust-flash"
wlx70f11c4bf34e: interface state UNINITIALIZED->ENABLED
wlx70f11c4bf34e: AP-ENABLED

minax007, 2022-04-17:
Dear all, 

@Raumzeit shard following information, which has resolved my connection issue with vtrust-flash:

https://github.com/ct-Open-Source/tuya-convert/issues/892#issuecomment-1100248239