ct-Open-Source/tuya-convert #553: Device partially update but seems to be stuck at intermediate firmware: Aoyococr X10S

linkages, 2020-02-24:
I am trying to flash the default tasmota.bin to a few Aoyococr X10S smart plugs. I got 2 of the 4 to work without any issues with 2 failing. 
Looking at the logs it seems that the intermediate firmware uploaded successfully but then the rest of the steps failed. I do not see an SSID of vflash-recovery on my network post the install.

I actually took one of the devices apart to see if I could put a header on it to manually flash tasmota on it but that is not going to work at all.

Is there anything I can do to recover from this?

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4247321/smarthack-mqtt.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4247322/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4247323/smarthack-web.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4247324/smarthack-psk.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4247325/smarthack-wifi.log)

kueblc, 2020-02-24:
Have you tried power cycling the device?

Assuming the intermediate firmware installed successfully, it should try to connect to `vtrust-flash` falling back to AP mode with `vtrust-recovery`. It will not respond to button presses in this state.

If it did not install successfully, it should roll back to stock firmware. It should respond to button presses as normal.

linkages, 2020-02-25:
I power cycled it by unplugging the plug from power for about 30 seconds and then plugging it back in. The button does not respond to anything. 

I re-ran start_flash.sh and it does not find the device.

When I plug in the device, sometime the led blinks blue slow 3 times and then turns off. Most of the time the led turns on blue and then turns off.

Can you tell from the logs if it actually flashed the intermediate firmware?
How long does it take for it to fail to AP mode?

axan2149, 2020-04-11:
I'm in the same boat. Did you ever get anywhere with this? plug bricked after initial flash, I don't think it ever got the intermediate firmware. Doesn't respond to button

kueblc, 2020-04-11:
@axan2149 If it doesn't respond to button, that means it is running the intermediate firmware

skpram219, 2020-05-05:
I had identical results as "linkages" above on two out of three devices.  The second one I tried completed although it could not save the old image.  The following is part of the output for the successful flash.
IoT-device is online with ip 10.42.42.42
Fetching firmware backup
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 1024k    0  1460    0     0     16      0 18:12:16  0:01:30 18:10:46     0
curl: Saved to filename 'firmware-b9ef4d.bin'
curl: (28) Operation timed out after 90000 milliseconds with 1460 out of 1048576 bytes received
Could not fetch a complete backup
Do you want to continue anyway? [y/N] y
======================================================
Getting Info from IoT-device
VTRUST-FLASH 1.5
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: b9ef4d
MAC: 84:0D:8E:B9:EF:4D
BootVersion: 7
BootMode: normal
FlashMode: 1M DOUT @ 40MHz
FlashChipId: 144068
FlashChipRealSize: 1024K
Active Userspace: user2 0x81000
======================================================
Ready to flash third party firmware!

skpram219, 2020-05-05:
I had identical results as "linkages" above on two out of three devices.  The second one I tried completed although it could not save the old image.  The following is part of the output for the successful flash.
"IoT-device is online with ip 10.42.42.42
Fetching firmware backup
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 1024k    0  1460    0     0     16      0 18:12:16  0:01:30 18:10:46     0
curl: Saved to filename 'firmware-b9ef4d.bin'
curl: (28) Operation timed out after 90000 milliseconds with 1460 out of 1048576 bytes received
Could not fetch a complete backup
Do you want to continue anyway? [y/N] y

Getting Info from IoT-device
VTRUST-FLASH 1.5
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: b9ef4d
MAC: 84:0D:8E:B9:EF:4D
BootVersion: 7
BootMode: normal
FlashMode: 1M DOUT @ 40MHz
FlashChipId: 144068
FlashChipRealSize: 1024K
Active Userspace: user2 0x81000

Ready to flash third party firmware!"

skpram219, 2020-05-06:
I now have four working Aoycocr  X10S.  This post describes how I recovered the two bricks mentioned in my previous post plus a third brick I made since then.  All bricked after running ./start_flash.sh.  I believe this recovery method gives a clue as to why the intermediate firmware is marginally compatible with the X10S.

First I removed the cover from the brick.  **At this point 120V is exposed so I never plugged it into the power outlet.**

I then carefully soldered wires on the 3.3V regulator chip.

My first try was to hook up the wires to two, 1.5V AA batteries (3V).  After literally hundreds of wire jiggles to make the esp8266 reboot in weird ways, somehow it started OK and reflashed.  This made me think that something about the startup of the esp chip with the intermediate firmware on it is the problem.

To see if the 3.3V supply was too slow getting to voltage for this firmware, I first added a 4,700 microfarad capacitor across the batteries and I then connected my second brick.  When I ran ./start_flash.sh , the brick immediately reflashed after doing a complete save of the old firmware, with no delays or unusual messages.

I used the capacitor method on my third brick with the same success.

There are long leads on the push button switch that must plug into a socket on the motherboard when reassembling.

The fact that the intermediate code does not run properly with a slow ramp up of the 3.3V supply, but the final code is fine, makes me think there might be a hint to a solution to make the intermediate firmware compatible with this hardware (a delay or brown-out reset).

linkages, 2020-05-06:
@skpram219 would you mind taking some pictures of which pins are the ones to use? Also, how did you seal the socket back up after you were done with it? I have one taken one of mine apart but it seems like it is basically broken. Did you hot glue it shut or something?

skpram219, 2020-05-06:
I was able to break the assembly joints on the case so it fit together perfectly.  I do not have a photo but the 3.3V regulator is shown.  "Output" is connected to +3V from my battery/capacitor.
[image]

iankaufmann, 2020-05-31:
I was stuck in the same spot where it would fail while attempting the backup after a few seconds.  

Commenting out line 92 (sleep 2) and then lines 98-104 from start_flash.sh so it just doesn't even attempt that part fixed 3 out of 4 of my plugs.  I would just start the script, power up the plug, and get through the options as fast as possible.

I decided to just try it without the backup rather than go down the road of opening these things up... and somehow it worked.  I was pretty surprised so I figured I'd share.  

1 of the 4 I am stuck on and it just blinks once (sometimes 2-4 times) then goes dead.  The button does nothing but I also don't get the "vtrust-recovery" AP or anything.  I wonder if that one died in the middle of installing the intermediate firmware or what... I just tried it about 20 times and I'm not getting anywhere with it.

MrPaulAR, 2020-06-04:
Of my 4
* Zero flashed properly.  All were stuck with the temporary firmware 
* One flashed with @iankaufmann suggestion for commenting
* Three flashed with @skpram219 suggestion for soldering

clharting, 2020-06-23:
I purchased 4 of these earlier this year, seemingly bricked the first two. After reading this thread, I was able to use @iankaufmann 's suggestion and got the third one working, but the fourth also seems to be bricked now... I was able to open one of the first two (without much trouble or significant damage), but my attempts to revive it haven't worked so far.
The question I have: if I apply 3.3V power to the voltage regulator pins, will that cause the unit to fall back to the factory firmware, then I just start over again with the modified start_flash.sh?

clharting, 2020-06-23:
I think I answered my own question... I applied power to the regulator pins, ran start_flash.sh without modification and it updated the firmware! One down, two to go...

n8henrie, 2020-06-23:
@clharting  @MrPaulAR  @skpram219  -- can any of you describe your strategy for opening it up without damaging the unit? I tried going around the seam on the back with a scalpel but that thing is stuck on there tight! Heat gun?

clharting, 2020-06-24:
@n8henrie , there are four tabs on the back panel. I used a utility knife to go along the long straight seams until the tabs popped. At that point, you can get a screwdriver in there and just work your way to one end, then the other. The shell flexes a bit,  but  I didn't break anything.
[image]

clharting, 2020-06-25:
Update: I have all 4 units working. I used Elmer's Glue to put them back together. Probably not a good long-term solution, but I don't expect to move these around a lot...

Mindsgoneawol, 2020-08-17:
> I was stuck in the same spot where it would fail while attempting the backup after a few seconds.
>
> Commenting out line 92 (sleep 2) and then lines 98-104 from start_flash.sh so it just doesn't even [...]

Could you post  the actual lines (text) or a screenshot please? I have 2 bricked and would like to try this but for some reason I am not understanding what those lines are suppose to be. Sleep 2 isn't on line 92 on mine for some reason. Thank you

iankaufmann, 2020-08-17:
Sure, it does look like the code has been updated since my post.  Looks
like it would currently be lines 97-110.

Basically the idea was that if you wait too long, the Aoyococr device
seemed to be turning off, going to sleep, or otherwise dropping the
connection.

So I just looked for what was holding it up.  Taking out the sleep 2 and
the backup process happened to work (once).  I still have a bricked one
that I can't get this to work on though.

I think it all comes down to exact timing... you have a few second window
to get it.  I figured, I am never going to use this plug as intended by the
manufacturer (on the cloud) so I don't care if I have a backup of the OG
firmware or not.  I'll throw it away bricked before I use it on the cloud.

Hope that helps.

[image: Screen Shot 2020-08-17 at 10.11.30 AM.png]

Mindsgoneawol, 2020-08-17:
Thank you. Going to dig out my rpi and see if i can get these 2 flashed. Hopefully it works so I can order more.

iankaufmann, 2020-08-17:
> Thank you. Going to dig out my rpi and see if i can get these 2 flashed. Hopefully it works so I can order more.

No prob.  For what it's worth, I had better luck with these:

BN-Link - BNC-60/U133TJ-2P 

100% success rate on those!  Still had a bit of a timing finesse to work through, but they all worked in a few attempts.  This was helpful:  https://donnlee.com/2020/07/10/tuya-convert-bn-link-bnc-60-u133tj-wifi-smart-plugs/

Not trying to shill the other product in the Aoyococr thread but I think it's relevant.  Maybe some of the tips in that article apply here too.  It's obvious that some devices have this issue with tyua-convert and timing.

The Gosund WP3 was by far the smoothest experience.  Flashed about 12 of them in a row without a single hitch.  But every energy monitoring plug I've tried has been a pain.

Good luck!

jrob, 2020-08-26:
The 3.3v trick worked for me. I bought 4 and 2 flashed fine and the other two went sideways. Thanks for the tip.

blade3609, 2020-10-09:
Thanks for the great tips! I previously bought 4 of these from amazon on 24th of april 2020, all 4 worked with tuya convert (had to try multiple times though). On the 26th of sep 2020 I bought 4 more, and all of these were stuck on intermediate firmware. 
Its listed that the X10S' are possibly firmware updated since 21st of april, and that they cant be converted due to new PSK format, but that's not the case from what ive seen.

the 3.3v trick worked for me, however i didn't have to use any capacitors nor did i have to solder. I just opened it up and connected jumper wires to ft232 uart and held them on the board while tuya convert did its work.

thanks again all who contributed.

zanchin, 2020-11-02:
@skpram219 Where did you connect the negative side of your battery? The ground pin of the regulator?

MrPaulAR, 2020-11-02:
Anything on the ground plane.