ct-Open-Source/tuya-convert #554: Phone won't stay connected to vtrust-flash AP results in 'Device did not appear with the intermediate firmware..."

EvadingRye, 2020-02-25:
Hi there,

Really love the project. I'm a newbie so please bear with me. I'm using Raspbian Buster Lite on my RPi4 to flash some smart devices, starting with a Teckin SP10 plug. I believe I'm using the newest Tuya Convert version. When paring my Android phone to the vtrust-flash wifi SSID, it connects but then when the Flashing begins it disconnects. I've attached my logs, hoping someone can spot the issue!

[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4251911/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4251912/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4251913/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4251914/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4251915/smarthack-psk.log)

kueblc, 2020-02-25:
Unfortunately this device is not supported, according to your web log:
```
WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware
```

closed 2020-02-25 by kueblc

EvadingRye, 2020-02-25:
Thank you for the quick response!