ct-Open-Source/tuya-convert #556: Teckin UK SP23 light doesn't stay on

mattydanby, 2020-02-29:
My SP23 was bought in late 2018 and Tasmota flash attempts v7.2.0-8.1.0 work but I'm unable to get the  button light to stay on when the toggle is on.

I have also changed the BlitzWolf SHP (45) GPIOs to the values stated here:  https://templates.blakadder.com/teckin_SP23.html

How can I now change to a v6 Tasmota version firmware or ESPurna firmware, I can't figure out how to now get the device into Pairing Mode again.

Thanks

tonysprenk, 2020-02-29:
Once the device is flashed (running Tasmota) tuya convert cannot be used anymore. Tasmota has an OTA update mechanism but you should ask about it over on the Tasmota github.

mattydanby, 2020-02-29:
I thought so, thanks.

closed 2020-02-29 by mattydanby