ct-Open-Source/tuya-convert #557: Force specific flash memory size

ccrisan, 2020-02-29:
Is there a way to instruct Tuya-Convert to alter the flash size specified by the boot partition?

I'll explain myself. I'm trying to flash a 2MB device with a custom firmware that boots and works with 1MB. My custom binary includes the first 512kB and, if written using esptool boots perfectly.

However, when supplied to Tuya-Convert, it uses the original boot header (in my case 2MB):

    e9 03 02 30

According to [this](https://github.com/espressif/esptool/wiki/Firmware-Image-Format), the `3` in last `30` indicates a 2MB board. If I manually change it to `20` (1MB board), the system boots fine.

When using 2MB, I noticed it keeps booting user2 zone, whereas it just flashed my custom firmware on the user1 zone. I assume it looks at the wrong address for the current user1/user2 configuration.

Any help would be appreciated.

ccrisan, 2020-02-29:
I found the answer myself and the answer appears to be positive.

After googling around a bit, I found the [source code](https://github.com/vtrust-de/esp8266-ota-flash-convert) of the helper that runs on the ESP after conversion. This piece of code has a function called `downloadRomToFlash` with an argument called `preserve_flash_settings`, which appears to be exactly what I need.

[This line](https://github.com/vtrust-de/esp8266-ota-flash-convert/blob/master/src/esp8266-ota-flash-convert.ino#L134) revealed that if one passes the `override` query argument to the HTTP request responsible of flashing the custom firmware, the custom firmware flash params will be used instead of the original ones.

To make Tuya-Convert pass this argument to ESP, I changed [a line](https://github.com/ct-Open-Source/tuya-convert/blob/master/scripts/firmware_picker.sh#L58) in `firmware_picker.sh` as follows (adding `&override=yes`):

    ...
    RESULT=$(curl -s "http://10.42.42.42/flash?url=http://10.42.42.1/files/$selection&override=yes") ||
    ...

It really works as expected, using the flash params I supply via my custom firmware.

closed 2020-02-29 by ccrisan

kueblc, 2020-02-29:
Happy to hear you were able to find and employ my undocumented feature. This is the first case I've heard of it being used, so I'm glad it was useful and worked as expected.

ccrisan, 2020-02-29:
@kueblc how do others deal with this particular problem? Because I feel I'm doing something wrong here.

I mean the `tasmota.bin` firmware bundled with Tuya-Convert, how does it work with all memory sizes available? How come it doesn't mess with the address where user1/user2 selection is stored? What is that address, by the way? Could you maybe please shed some light on this matter or point me to some docs?

kueblc, 2020-02-29:
I haven't heard of anyone else with a 2MB device. AFAIK Tuya modules were all 1MB.

Normally in devices >= 1MB, `user1` is at `0x1000` and `user2` is at `0x81000`. This is documented by Espressif.

After installing Tasmota, `user1` and `user2` are meaningless since the Arduino platform uses it's own partition layout.

ccrisan, 2020-02-29:
> I haven't heard of anyone else with a 2MB device. AFAIK Tuya modules were all 1MB.
 
Actually my device is a Blitzwolf SS3 smart switch (apologies for not mentioning it so far), which appears to carry 2MB of flash.

> Normally in devices >= 1MB, `user1` is at `0x1000` and `user2` is at `0x81000`. This is documented by Espressif.

My custom firmware uses those addresses too (following Espressif's recommendations), regardless of the actual flash size. Nevertheless, the device somehow stays into user2, while it should boot user1. Do you have any idea where this particular flag is stored in the flash?

eku, 2020-03-01:
@kueblc 
> I haven't heard of anyone else with a 2MB device. AFAIK Tuya modules were all 1MB.

Well, that's not always true. Those IR devices are usually 2MB devices, so is mine

```
ChipID: fa8a4c
MAC: xxxxx
BootVersion: 7
BootMode: normal
FlashMode: 1M QIO @ 40MHz
FlashChipId: 1540c8
FlashChipRealSize: 2048K
Active Userspace: user2 0x81000
```