ct-Open-Source/tuya-convert #559: Device did not appear with the intermediate firmware - MoesHouse Smart Downlight

leifclaesson, 2020-03-03:
This is the device:
[https://www.aliexpress.com/item/4000296147883.html]

I'm in the process of researching what lights to get for the house I'm building. This was the first one to arrive. I've previously successfully flashed several Tuya devices from my RPi 3B+ with Raspbian Buster so I'm confident in the methodology.

This MoesHouse downlight has never been connected to a SmartLife app, it is a brand new device. So, I do not know what firmware version is on it.

I place it into EZ config mode (fast flashing), then run start_flash, at which point the downlight _stops_ flashing and tuya_convert simply keeps resending SmartConfig packets until it fails out with the "Device did not appear with the intermediate firmware" option.

I've seen a few other threads here with this issue, where the conclusion was "not an ESP8266"... but in my case, I took the device apart, and it IS an ESP8266. So, that's not it.
Here's what it looks like inside.

[image]

Here are the log files:
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4282496/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4282498/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4282499/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4282500/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4282501/smarthack-wifi.log)

I also followed keublc's instructions in another thread (issue 472) and did a tcpdump capture like so:
sudo tcpdump -i wlan0 -w capture.pcap

Here it is:
[capture.zip](https://github.com/ct-Open-Source/tuya-convert/files/4282507/capture.zip)

And finally.. with all those nice soldering islands, I soldered on some wires and hooked up my serial jig, and with esptool I saved the original firmware!

[moeshouse_downlight_1m.zip](https://github.com/ct-Open-Source/tuya-convert/files/4282511/moeshouse_downlight_1m.zip)

I did _not_ manually load tasmota on it, because I'm going to need a hundred of these things for the house.. soldering is not an option at this scale. so it's still a virgin device as far as the software is concerned.

Any help greatly appreciated, let me know if I can provide any more information.
Best regards,
Leif Claesson

kueblc, 2020-03-03:
Unfortunately, from your psk log, it looks like you ended up with the latest patched firmware from Tuya. Follow #483 for more information and hopefully an eventual resolution.

kueblc, 2020-03-03:
Also, thank you for your diligence in collecting data and downloading the firmware for analysis! This will certainly help with the effort to bring custom firmware to these devices.

closed 2020-03-03 by kueblc

leifclaesson, 2020-03-03:
Thanks, kueblc! I just discovered #483 while searching for DECRYPTION_FAILED_OR_BAD_RECORD_MAC which I found in the log, after making this post. Good thing I didn't see it at first or I may not have bothered with the soldering iron :).

jeylites, 2020-09-09:
@leifclaesson 

Hi, just checking in. I'm planning to get a few of the Moes downlights, have you had any luck getting tasmota installed?

Also, how is the warm white and white colors compared to traditional downlights?