ct-Open-Source/tuya-convert #560: In search for original Teckin SP22 firmware

IRp3r, 2020-03-03:
Hello,

sorry if this isn't the right place.
I need to reflash the original firmware on one (about a year old, should use ESP8266) Teckin SP22.
Maybe someone from the community still has the original firmware image from the setup process.
Would be really cool!

If this request does not fit here, just let me know and I'll remove it.

Kind Regards

kueblc, 2020-03-03:
You won't be able to use another device's firmware image with yours if you wish to use the Tuya cloud. Each firmware image has a uniquely assigned key specific to each device.

IRp3r, 2020-03-04:
@kueblc  alright, thanks for your reply. I hope I'll find the original firmware somewhere (I know I extracted it somewhere). And even if not, luckily they are cheap.

closed 2020-03-04 by IRp3r