ct-Open-Source/tuya-convert #562: vtrust-flash does not show up

Aryankids, 2020-03-05:
Have cloned the latest repo on a raspbian buster lite (Raspberry 3 B+), installed pre-reqs, started the flash but the AP does not appear.  _sudo wpa_cli_ terminate  has no effect and get more or less the same log back:   

The log says:

System info
===========
31856ad
Linux raspberrypi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
OpenSSL 1.1.1d  10 Sep 2019
Dnsmasq version 2.80  Copyright (c) 2000-2018 Simon Kelley
Compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile

This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.
hostapd v2.8-devel
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2019, Jouni Malinen <j@w1.fi> and contributors
Python 3.7.3
===========
Attempting to stop wpa_supplicant
Configuring AP interface...
RTNETLINK answers: Operation not possible due to RF-kill
Error: Device for nexthop is not up.
Error: Device for nexthop is not up.
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
rfkill: WLAN soft blocked
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:a7:95:84 and ssid "vtrust-flash"
Failed to set beacon parameters
wlan0: Could not connect to kernel driver
Interface initialization failed
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: Unable to setup interface.
wlan0: interface state DISABLED->DISABLED
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface wlan0 wasn't started
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...

==> smarthack-web.log <==
Listening on 10.42.42.1:80

==> smarthack-mqtt.log <==
1583368816: mosquitto version 1.5.7 starting
1583368816: Using default config.
1583368816: Opening ipv4 listen socket on port 1883.
1583368816: Opening ipv6 listen socket on port 1883.

==> smarthack-udp.log <==
Listening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
192.168.1.26 {"ip":"192.168.1.26","gwId":"05200414dc4f22c50331","active":2,"ability":0,"mode":0,"encrypt":true,"productKey":"MyProductKey","version":"3.1"}
192.168.1.13 {"ip":"192.168.1.13","gwId":"205513782cf432191189","active":2,"ability":0,"mode":0,"encrypt":true,"productKey":"MyProductKey","version":"3.3"}
192.168.1.25 {"ip":"192.168.1.25","gwId":"63630075dc4f22d5e112","active":2,"ability":0,"mode":0,"encrypt":true,"productKey":"MyProductKey","version":"3.1"}

-----------

Also the script output says:

tuya-convert v2.4.3
Checking for network interface wlan0... Found.
Checking UDP port 53... Occupied by dnsmasq with PID 513.
Port 53 is needed to resolve DNS queries
Do you wish to terminate dnsmasq? [y/N] y
Attempting to stop dnsmasq.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Occupied by mosquitto with PID 499.
Port 1883 is needed to run MQTT
Do you wish to terminate mosquitto? [y/N] y
Attempting to stop mosquitto.service
Checking TCP port 8886... Available.
sudo: ufw: command not found
======================================================
  Starting AP in a screen.
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

Any help would be hugely appreciated 
regs

kueblc, 2020-03-05:
> rfkill: WLAN soft blocked

Try `rfkill unblock wifi`

Aryankids, 2020-03-07:
Thank you so much for the quick reply. That did the trick.

closed 2020-03-07 by Aryankids