ct-Open-Source/tuya-convert #563: How to convert this smart plug Aunics IT

pex81, 2020-03-05:
**On the website https://templates.blakadder.com/aunics_it.html I found the template, but unfortunately I can't flash this socket.
I have tried several times without success**
_I attach the logs of tuya-converter!_
Can anyone help me?

```
root@raspberrypi:/home/pi/Desktop/pacchetti/tuya-convert # ./start_flash.sh 
tuya-convert v2.4.3
======================================================
TUYA-CONVERT

https://github.com/ct-Open-Source/tuya-convert
TUYA-CONVERT was developed by Michael Steigerwald from the IT security company VTRUST (https://www.vtrust.de/) in collaboration with the techjournalists Merlin Schumacher, Pina Merkert, Andrijan Moecker and Jan Mahn at c't Magazine. (https://www.ct.de/)

======================================================
PLEASE READ THIS CAREFULLY!
======================================================
TUYA-CONVERT creates a fake update server environment for ESP8266/85 based tuya devices. It enables you to backup your devices firmware and upload an alternative one (e.g. ESPEasy, Tasmota, Espurna) without the need to open the device and solder a serial connection (OTA, Over-the-air).
Please make sure that you understand the consequences of flashing an alternative firmware, since you might lose functionality!

Flashing an alternative firmware can cause unexpected device behavior and/or render the device unusable. Be aware that you do use this software at YOUR OWN RISK! Please acknowledge that VTRUST and c't Magazine (or Heise Medien GmbH & Co. KG) CAN NOT be held accountable for ANY DAMAGE or LOSS OF FUNCTIONALITY by typing yes + Enter

yes
Checking for network interface wlan0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
sudo: ufw: comando non trovato
======================================================
  Starting AP in a screen....
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
..............
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] N
======================================================
Cleaning up...
Closing AP
Exiting...
```

[tuya-converter.log](https://github.com/ct-Open-Source/tuya-convert/files/4295294/tuya-converter.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4295265/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4295266/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4295267/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4295269/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4295271/smarthack-wifi.log)

kueblc, 2020-03-06:
Unfortunately, according to your psk log, you have a device with the latest patched firmware from Tuya. We do not currently have support for this firmware. Please follow #483 for more information.

closed 2020-03-06 by kueblc

pex81, 2020-03-06:
doing some network diagnostics, I noticed that the udp 6668 port is open, can you try something?

kueblc, 2020-03-06:
You can use `tuyapi` to control the stock firmware locally, or you can open the device and flash third party firmware directly, these are your only options at this point.

fireport, 2020-05-02:
@pex81 I have the same problem. Did you found a solution? Did you open the plug case to flash with a serial interface?
Thank you

pigr8, 2020-08-20:
flashed succesfully with 8.4.0, cannot be done over OTA 'cause comes shipped with the fixed tuya firmware, it has to be done soldering and flashing via serial.. it's a common tywe2s (esp8285 with 1Mb flash?) so it's easy.

you need to open it to do that (also easy, it has a light ammount of glue and 4 clips), almost no damage to be seen if done carefully.

marcolino7, 2020-11-15:
@pigr8 
can you provide more detail on how procede for flashing soldering serial adapter?
Thanks

pigr8, 2020-11-15:
If you open the plug you can see that the pins of the esp8285 are easily accessible and you can solder 4 wires onto the correct pins (just search tywe2s pinout on google and to find the correct ones: 3v3 gnd rx tx).. using a cheap serial usb adapter just connect the 4 wires and flash tasmota the common way (es: tasmotizer).. after that test the plug, desolder the wire and you can close everything up.

bare in mind that someone reported that there is a capacitor between rx and gnd pins, you should be able to easily remove it to be able to flash tasmota.

cheers