ct-Open-Source/tuya-convert #564: flash the deltaco sh-p02 does not work

farbrot, 2020-03-06:
Hi
I saw another post about this but that one was closed. So as a newbie I repost since I am a newbie on this matter. :)
Tried to flash this sh-p02 without any success. 
Looked in the logs and really don't understand which one to look in.
Is there a tutorial on the matter that I can find. 
Thanks for all help.

kueblc, 2020-03-06:
Please share some information so we can help you debug. I will need to see the logs found in your `tuya-convert/scripts` or `tuya-convert/backups` directory.

farbrot, 2020-03-08:
Hi,
I starting to think it is the problem with the new firmware #483?? 
I'm not to sure its the issue.

[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4303126/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4303127/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4303128/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4303129/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4303130/smarthack-psk.log)

leifclaesson, 2020-03-08:
Not the same issue, farbrot. The real error message is in the UDP log:
**WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware**

The MAC address appears to be from "Hangzhou Hangzhou Aixiangji Technology" -- I don't recognize this so I don't know what microcontroller it is, but if it's not an ESP82xx then Tasmota/ESPeasy won't work, unfortunately.

kueblc, 2020-03-08:
According to your web log
```
WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware
```
We only support ESP based devices

closed 2020-03-08 by kueblc

farbrot, 2020-03-08:
[image]

Just show the device

farbrot, 2020-03-08:
> Not the same issue, farbrot. The real error message is in the UDP log:
> **WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware**
>
> The MAC address [...]

It the same image as the show at the supported devices in https://templates.blakadder.com/deltaco_SH-P02.html

thrawnarn, 2020-03-09:
I had the same problem as @farbrot . Same error message. I decided to sacrifice my plug to know if there is something to do. And it seams like Deltaco has change chipset inside of them. At least I couldn't find an ESP.

[image]
[image]
[image]

meingraham, 2020-03-09:
Yup, looks like they have switched to the RTL Wi-Fi chip.

lerra, 2020-04-18:
Any chance to get support for the new chipset?