ct-Open-Source/tuya-convert #567: Gosund Switch dead after flash

tgmm82, 2020-03-07:
I did my first OTA flash of tasmota today and i fear I bricked my device.  Everything seemed to go find but now it does not work.  

This is my first tasmota flash and I am very much a newbie to using Python as well.  I used a pi 3 to flash the device and all was successful, I was able to connect to the device's WiFi and go to the screen to set my home network credentials.  I scanned the WiFi, found my network, entered my password and hit save. 

Then nothing.  

My switch shows no power at all.  I cannot get nay response from it at all.  It is not showing on my network so I cannot connect to it so I cannot connect to the tasmota settings.  

I have no control on the device at all.  When I press the switch nothing happens, there are no lights flashing or appearing at all.  I have pressed and held the switch to try to reset it and no response.  

I fear it is bricked but I have no experience here.  

I apologize, I am not sure how to find any logs to show for this.

leifclaesson, 2020-03-07:
Just guessing here -- what if you entered the wrong WiFi password?
You definitely got Tasmota flashed alright or you would not have been able to connect to the device's WiFi.

Have you tried the following Tasmota reset procedure?
[https://github.com/arendst/Tasmota/wiki/Fast-power-cycle-device-recovery](https://github.com/arendst/Tasmota/wiki/Fast-power-cycle-device-recovery)

It's worth a try.

kueblc, 2020-03-07:
This is not a tuya-convert issue, as you have stated Tasmota was flashed successfully. I would agree with Leif's advice, failing this, get in touch with Tasmota discord support. Good luck!

closed 2020-03-07 by kueblc