ct-Open-Source/tuya-convert #570: Docker issue, possibly "address already in use"

picto3000, 2020-03-17:
Any help appreciated. I'm seeing DNSMASQ errors via docker. I do have other things running in Docker so there's possibly a port conflict.

smarthack-wifi.log
```
System info
===========
31856ad
Linux Burbank 5.3.0-40-generic #32-Ubuntu SMP Fri Jan 31 20:24:34 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
OpenSSL 1.1.1  11 Sep 2018
Dnsmasq version 2.79  Copyright (c) 2000-2018 Simon Kelley
Compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify

This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.
hostapd v2.6
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2016, Jouni Malinen <j@w1.fi> and contributors
Python 3.6.9
===========
Configuring AP interface...
RTNETLINK answers: File exists
RTNETLINK answers: File exists
Starting DNSMASQ server...

dnsmasq: failed to create listening socket for 10.42.42.1: Address already in use
Starting AP on wlp0s20f3...
Configuration file: /dev/stdin
Using interface wlp0s20f3 with hwaddr 04:d3:b0:15:26:38 and ssid "vtrust-flash"
wlp0s20f3: interface state UNINITIALIZED->ENABLED
wlp0s20f3: AP-ENABLED 
wlp0s20f3: AP-STA-CONNECTED d8:f1:5b:92:fa:cf
wlp0s20f3: AP-STA-DISCONNECTED d8:f1:5b:92:fa:cf
wlp0s20f3: AP-STA-CONNECTED d8:f1:5b:92:fa:cf
wlp0s20f3: AP-STA-CONNECTED f4:5c:89:79:d2:eb
wlp0s20f3: AP-STA-DISCONNECTED d8:f1:5b:92:fa:cf
wlp0s20f3: AP-STA-CONNECTED d8:f1:5b:92:fa:cf
wlp0s20f3: AP-STA-DISCONNECTED d8:f1:5b:92:fa:cf
wlp0s20f3: AP-STA-CONNECTED d8:f1:5b:92:fa:cf
wlp0s20f3: AP-STA-CONNECTED b8:5d:0a:67:92:c4
wlp0s20f3: AP-STA-DISCONNECTED d8:f1:5b:92:fa:cf
wlp0s20f3: AP-STA-CONNECTED d8:f1:5b:92:fa:cf
wlp0s20f3: AP-STA-DISCONNECTED b8:5d:0a:67:92:c4
wlp0s20f3: AP-STA-DISCONNECTED f4:5c:89:79:d2:eb
wlp0s20f3: AP-STA-DISCONNECTED d8:f1:5b:92:fa:cf
wlp0s20f3: AP-STA-CONNECTED d8:f1:5b:92:fa:cf
wlp0s20f3: AP-STA-CONNECTED b8:5d:0a:67:92:c4
wlp0s20f3: AP-STA-DISCONNECTED b8:5d:0a:67:92:c4
```
smarthack-udp.log
```
Listening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
```

smarthack-psk.log
```
Traceback (most recent call last):
  File "./psk-frontend.py", line 113, in <module>
    main()
  File "./psk-frontend.py", line 99, in main
    proxies = [PskFrontend(gateway, 443, gateway, 80), PskFrontend(gateway, 8886, gateway, 1883)]
  File "./psk-frontend.py", line 46, in __init__
    self.server_sock = listener(listening_host, listening_port)
  File "./psk-frontend.py", line 17, in listener
    sock.bind((host, port))
OSError: [Errno 98] Address already in use
```

smarthack-mqtt.log
```
1584422260: mosquitto version 1.4.15 (build date Tue, 18 Jun 2019 11:42:22 -0300) starting
1584422260: Using default config.
1584422260: Opening ipv4 listen socket on port 1883.
1584422260: Error: Address already in use

```

smarthack-web.log
```
Could not start server on port 80
Close the process on this port and try again
```

kueblc, 2020-05-21:
Hi @picto3000 have you resolved your issue? Make sure the required ports are available

picto3000, 2020-05-21:
I couldn't resolve it. From memory, as my Docker NUC also runs pihole I couldn't just install it on the NUC's Linux distro as port 53 etc was already in use. I ended up dusting off an old laptop and loaded up a Lubuntu Live CD. Not perfect as I don't have a flashing env setup and ready to go at a moment's notice.

Here's my Docker-Compose for completeness:
```
# # Tuya-Convert - https://github.com/ct-Open-Source/tuya-convert
# # ./usr/bin/tuya-convert/start_flash.sh
   tuya:
     container_name: tuya-convert
     image: tuya:latest
     privileged: true
     network_mode: host
     restart: no
     environment:
       - WLAN=wlp0s20f3 #wlan0
       - AP=vtrust-flash
       - PASS=flashmeifyoucan
       - GATEWAY=10.42.42.1
     volumes:
       - ${USERDIR}/docker/tuya-convert/backups:/usr/bin/tuya-convert/backups
       - ${USERDIR}/docker/tuya-convert:/usr/bin/tuya-convert
```

tablatronix, 2020-11-26:
Anyone figure this out?
[image]

jenniferlee1818, 2021-12-17:
same?
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/7737538/smarthack-wifi.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/7737539/smarthack-web.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/7737540/smarthack-psk.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/7737541/smarthack-mqtt.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/7737542/smarthack-udp.log)

clmcavaney, 2023-07-05:
Not sure if this relates, but see https://github.com/ct-Open-Source/tuya-convert/issues/1040#issuecomment-1620868898
I found I had a conflict with a service running on the host.  As the docker configuration is using host networking, there was a conflict.