ct-Open-Source/tuya-convert #571: Flashing Aisirer AWP07L Smartplug failed two times

pilotseye, 2020-03-21:
**Did you have issues getting a device to flash?**
Aisirer AWP07L Smartplug, flashing failed after upgrade.bin and devices seem to be bricked now.

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4363390/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4363391/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4363392/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4363393/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4363394/smarthack-wifi.log)

kueblc, 2020-03-22:
Restart `start_flash.sh` and when it prompts you to connect a phone and put the device in pairing mode, don't. Just plug the device back in and press enter.

pilotseye, 2020-03-22:
Thanks, I gave it a try, but it still fails, the devices seem to be in pairing mode, but won't connect.

bogorad, 2020-07-06:
Same here. Oh well, will have to solder :(