ct-Open-Source/tuya-convert #572: TW-02-based (WinnerMicro W600-based) smart socket

reivilibre, 2020-03-23:
I have a Tuya-based smart socket which unfortunately turned out not to be ESP8266-based (woops, I didn't realise they existed and so didn't pay much attention when ordering).

I cracked it open.

Turns out it is based on a ESP-alike 'TW-02' module which uses the ARM-based W600.
W600 seems to have MicroPython ports and so on — I believe it should be possible to find/make firmware images for it if it comes to it.

So my question now would be — can we get tuya-convert to flash it?

I don't see anything in the code that would fundamentally stop an attempt to flash an ESP image to a non-ESP board, though warnings are produced.

However, it seems to me that the same API call is being made again and again — see logs, it feels as though it isn't progressing and even if I had a W600-compatible firmware image, it wouldn't get around to using it.

I have never seen tuya-convert against an ESP device and don't know where to start with this endeavour – if it is at all possible.

Do you have any more insight?

kueblc, 2020-03-23:
Unfortunately I do not foresee this being supported anytime soon. The hack only works with ESP based hardware with compatible Tuya firmware.

There is nothing in the code stopping `tuya-convert` from continuing the hack -- it simply does not work with non ESP devices. This happens to be a Good Thing since uploading ESP firmware to a non ESP device would have a high chance of bricking the device.

Warnings were added to help users understand why `tuya-convert` is not working for any given device. They do not alter the program control flow.

I do not have any of these WinnerMicro based devices or I might try to develop a hack for them too. It would also obviously need an alternative firmware as you mentioned, and I'm not familiar with any mature projects attempting to do this.

If the situation changes I would be happy to revisit this. Closing as addressed, but comments left open for any new developments.

closed 2020-03-23 by kueblc

reivilibre, 2020-03-24:
My confusion is: what specifically makes this hack only work for ESP devices?

My understanding is that the hack 'just' tricks the device into performing an OTA firmware upgrade with an image of our choosing; if I have misunderstood or underestimated the effort you have made, I'm sorry :).

If you are interested enough and it is just a matter of having a device, I could be happy to send you one… (though it is a UK socket which may be totally useless to you depending on where you live!)

In any case, I have one cracked open (the other will follow if I meet any success) on my desk now and, when I get a real 3.3V USB serial converter (my current one killed a Sonoff as the logic level emitted is still 5V in 3.3V mode :man_facepalming:!), I will see if I can reach any kind of bootloader to reflash it.

I notice it doesn't react to button presses for ~5 seconds after plugging in, maybe there is a bootloader giving a chance to flash there.

If I can get that far, I will try and create a basic W600 firmware which is Tasmota-like.

Pardon my manners, I forgot to thank you for this project in my prior post :). All the best.

kueblc, 2020-03-24:
> My confusion is: what specifically makes this hack only work for ESP devices?

This is a great question and I would be thrilled if you find any answers. [Tuya has separate SDKs for each of their supported platforms](https://github.com/TuyaInc). This project was built by reverse engineering the upgrade process for ESP devices, which apparently must be different for other devices/SDKs or else it should work for them too. Without access to other devices like this, we can't know what they do differently.

Another point is the accessibility of the ESP8266. The SDK is mature, hardware cheap, and alternative firmware projects thriving. It was, and still is in many places, the dominate microcontroller powering IoT.

> If you are interested enough and it is just a matter of having a device, I could be happy to send you one… (though it is a UK socket which may be totally useless to you depending on where you live!)

Thank you, kind of you to offer, I am in the US though.

> If I can get that far, I will try and create a basic W600 firmware which is Tasmota-like.

I am looking forward to hearing your progress. If you develop a hack for this platform too I would gladly accept a PR. We'll need some way of making sure only compatible binaries are sent if we start supporting multiple platforms.

TRSx80, 2020-06-22:
Hai guise,

I found this thread by way of adventure I outlined in post linked immediately above this one.

Turns out I also have the TW-02 module.

@kueblc,

I am in US and would be happy to send you one (or more) as I had originally bought a set of 4.

Let me know what you think!

Cheers!

TRSx80, 2020-06-22:
@reivilibre,

There is https://w600.chip.haus which I learned about in my earlier post where I first learned about this device.

I never even so much as flashed Tasmota yet, but I suppose there might be enough information over at that site a noob like me to begin poking around, at least...

I will share whatever I find, of course.

kueblc, 2020-06-22:
I'm curious if you're able to get an alternative firmware running on the W600, such as [MicroPython](https://docs.wemos.cc/en/latest/tutorials/w600/get_started_with_micropython_w600.html)

TRSx80, 2020-06-22:
Well, I am low (maybe approaching medium by now?) level wizard, with some tools and free time, so certainly willing to give it a shot.

I already started doing some research on the Internet, but any pointers to info on how to get started would be appreciated (save me a little time maybe).

TRSx80, 2020-06-22:
I see `w600tool` only looks like it uploads firmware.  My first thought is to try and dump the existing firmware...

I mean, I have 4 of them, but still...

reivilibre, 2020-06-22:
Sorry for not keeping you updated, but *yes* I did manage to get MicroPython running on the W600 in a TW-02 round smart socket.

It has VCC, GND, TX, RX pins clearly marked (and findable on a pinout sheet, I can fetch for you if you'd like).

In my case, I had to desolder the tabs attached to the male (plug) end of the smart socket to get access to the pins of the TW-02 (don't know how similar the US variant looks).

The RST pin of the TW-02, which I found absolutely necessary to touch to GND when trying to flash with w600tool, was quite hard to get to, but I could poke it with the male end of a DuPont connector as and when needed, which was good enough for me.

On my model, the push button switch is shared with the (RX?) pin of the TW-02 and so it is inconvenient, but I'm still working on it as and when I have time.

Hope this may be useful to you, if you have questions let me know, @TRSx80 .

TRSx80, 2020-06-22:
> VCC, GND, TX, RX pins clearly marked

Yes they are clearly marked on mine as well (can be seen in teardown link above).

Thanks a lot for the other tips!  Sure to save me some time...

So, what can we do once we get MicroPython working?  Sorry if that's a dumb question, this is all new to me here.

I still think the first thing we should be doing is to dump the existing firmware, no?

TRSx80, 2020-06-22:
@reivilibre must be better at soldering than me.  I ended up desoldering the pads on the main board while trying to attach wires to them.  :cry: 

So I just desoldered the entire module and then it was easier to get some wires onto it.

Got the following serial output (not much):
```
[01-01 00:00:10 TUYA Notice][tuya_device.c:694] free_mem_size:89888
[01-01 00:00:20 TUYA Notice][tuya_device.c:694] free_mem_size:89888
[01-01 00:00:30 TUYA Notice][tuya_device.c:694] free_mem_size:89888
[01-01 00:00:40 TUYA Notice][tuya_device.c:694] free_mem_size:89888
[01-01 00:00:50 TUYA Notice][tuya_device.c:694] free_mem_size:89888
[01-01 00:01:00 TUYA Notice][tuya_device.c:694] free_mem_size:89888
[01-01 00:01:10 TUYA Notice][tuya_device.c:694] free_mem_size:89888

< TUYA IOT SDK V:2.0.0 BS:30.01_PT:2.2_LAN:3.3_CAD:1.0.1_CD:1.0.0 >
[FW]: TC0025_TW02_PLUG_L1_P0 | [HW]: TW-02 V2.0 | [SW]: V1.4.0 (Nov 18 2019 08:48:17) | [PID]: nqvhejakb112obkn
[01-01 00:00:00 TUYA Notice][hw_table.c:532] wifi status is :0
[01-01 00:00:04 TUYA Notice][tuya_main.c:143] wf_assign_ap_scan failed(-606)
[01-01 00:00:04 TUYA Notice][gw_intf.c:2270] gw_cntl.gw_wsm.stat:1
[01-01 00:00:04 TUYA Info][wifi_hwl.c:1261] Low Power Mode
[01-01 00:00:05 TUYA Notice][hw_table.c:532] wifi status is :0
[01-01 00:00:14 TUYA Notice][tuya_device.c:694] free_mem_size:89904
[01-01 00:00:24 TUYA Notice][tuya_device.c:694] free_mem_size:89904
[01-01 00:00:34 TUYA Notice][tuya_device.c:694] free_mem_size:89904
[01-01 00:00:44 TUYA Notice][tuya_device.c:694] free_mem_size:89904
[01-01 00:00:54 TUYA Notice][tuya_device.c:694] free_mem_size:89904
```
Gap in middle there is where I jumped RST to GND to reset it and get logs from the beginning.  Serial runs at 115200 apparently (lucky guess, same as I had last set `pio device monitor` for).

Because you basically have to cut into the case in order to get at the headers, what I am thinking I will do will be to try and sandbox them using [Mock Tuya Cloud](https://github.com/kueblc/mocktuyacloud) and just leave the stock firmware alone.

I do still have this one module out though, willing to keep playing around with this one...

reivilibre, 2020-06-23:
I don't know how to dump the current firmware (I don't know if it is allowed by the bootloader, which afaict is called 'secboot' which sounds like they may have gone for 'secure').

My current approach is to write some MicroPython code to connect to an MQTT server and drive the relay that way (it's really easy to do as it's just flicking a GPIO).

I'll be happy to share code once it's ready — no guarantees on when this would be, though, sorry.

TRSx80, 2020-06-23:
Yeah, shouldn't be too hard to figure out, as you say just flicking a GPIO.  Might help others though, no rush from my end.  For me this was all just for fun / exploration / science / advancing the state of the art, if you will...

I don't know how your plugs are laid out across the pond (I have seen some pics, not sure if yours is same) but in order to get at the relevant pins on the US model, you basically need to cut the back of the case.  Maybe someone else is better at disassembly than me though (I posted a lot of pics in my disassembly thread if anyone has ideas).

But for me, trying to get my other (3) units working, due to having to cut the case I think I will give up on flashing alternative firmware for now, and instead try and sandbox them using something like tuyapi or maybe Mock Tuya Cloud, etc. plus maybe some firewall rules, separate subnet, or whatever, and just try to use them in containment, but with stock firmware.

Thrasher2020, 2020-06-25:
I've got two of these - I did wonder why they wouldn't flash. Opened them up to find the TW-02 V2 board :(

TRSx80, 2020-06-25:
> Opened them up

I hope you did the non-destructive way as shown [at the end of my teardown thread](https://github.com/ct-Open-Source/tuya-convert/issues/680#issuecomment-649611485), unlike me who sacrificed a module on the altar of science, so that others could be spared the same fate!  :smile:

johanson, 2020-06-27:
> @Thrasher2020,
> 
> > Opened them up
> 
> I hope you did the non-destructive way as shown [at the end of my teardown thread](https://github.com/ct-Open-Source/tuya-convert/issues/680#issuecomment-649611485), unlike me who sacrificed a module on the altar of science, so that others could be spared the same fate!

The EU/Schuko ones are glued as well but (probably because of the shape) it's possible to open them leaving them in semi-ok condition :) Mine's TW-02 V2 as well, looks like some older models were ESP8266 based, hence people reporting to successfully flashing them.

TRSx80, 2020-07-02:
> I'm curious if you're able to get an alternative firmware running on the W600, such as [MicroPython](https://docs.wemos.cc/en/latest/tutorials/w600/get_started_with_micropython_w600.html)

@kueblc,

As I was sitting on the throne this morning, I had a thought.  If we are able to get some MicroPython based firmware working, then this opens up the possibility for eventually possibly doing FOTA update with `tuya-convert`?  With our newly developed firmware?  Is this what you were hinting at further up thread?

At that point, as you said earlier I think, it would become necessary to add some checks to make sure which device we are dealing with, but I suppose that is a tangential discussion to the main issue here.

kenthinson, 2020-07-16:
more info https://fccid.io/2ASQV-TW-02/User-Manual/User-manual-4211029.pdf

p0ns, 2020-07-20:
https://docs.w600.fun/?p=arduino/start.md seems like it supports arduino

p0ns, 2020-07-20:
https://docs.w600.fun/?p=product/tw-02.md 
https://docs.w600.fun/?p=arduino/start.md

seems like it supports Arduino

ceaswaran, 2020-08-02:
Has anyone explored to the possibility of replacing TW-02  with ESP01 or any other esp8266 board so we can tuyaconvert/tasmotize this?

sigma-guilhfer, 2020-08-27:
http://www.winnermicro.com/en/html/1/156/158/497.html

I found this.
But the effort to crack it is not worthy, in my opinion.
Maybe the suggestion from @ceaswaran is better, but at this point I already gave up on this topic

ceaswaran, 2020-08-27:
Managed to remove the TW02 module & Soldered Wemos D1 Mini
I was able to map the Pins & Get all parameters working . Even the Power Monitoring works great.
[image]
[image]

TRSx80, 2020-08-27:
> But the effort to crack it is not worthy, in my opinion.

In the case of [my particular device](https://github.com/ct-Open-Source/tuya-convert/issues/680#issuecomment-649611485) (already mentioned up thread), I agree, as I don't even see how you could physically get at the leads without destroying the nice small form factor of the physical device, which would totally defeat the purpose IMO.

However, something else I said up thread seems to (so far) being overlooked:

>> I'm curious if you're able to get an alternative firmware running on the W600, such as MicroPython
>
>@kueblc,
>
> [...] I had a thought. If we are able to get some MicroPython based firmware working, then this opens up the possibility for eventually possibly doing FOTA update with tuya-convert? With our newly developed firmware? Is this what you were hinting at further up thread?

I think this should be the way forward, as (after initial development) could potentially be used to flash devices over the air.  And thus not destroying the (nice, compact, neat) physical form factor in the process.

Unless I am misunderstanding something?

sigma-guilhfer, 2020-09-10:
I was able to do the same. Here is the GPIOS I used:
[image]

(I had problems using this model, some components was inverted https://templates.blakadder.com/ZBR-001.html)

I used a ESP12F and followed the instructions from this guy: https://www.youtube.com/watch?v=37KlzYhJaQs

Also the instructions from Tasmota itself: https://tasmota.github.io/docs/Getting-Started/

Hope it can help someone

lroubinov, 2020-09-27:
@sigma-guilhfer or @ceaswaran 
can you explain which pin you soldered to the wemos d1
in the picture attached it's hard to follow
thank you

ceaswaran, 2020-09-27:
> @sigma-guilhfer or @ceaswaran
> can you explain which pin you soldered to the wemos d1
> in the picture attached it's hard to follow
> TX on board not soldered?
> Where you soldered the on board RX in the WEMOS?
> GND on board not soldered?
> 
> thank you

This is what I did.
Step 1. Flashed D1 mini with Tasmota & Changed it to the generic module.
Step 2. with socket disconnected from Mains power !!! Put the multimeter in Continuity mode & probed the pins that lead to TW-02 pads) Example: Probing pin of BL0937 (CF) & pads of microcontroller one by one whichever beeps I noted it as CF. Similarly all.

Bacically u need the following 6 GPIO for this plug to work.
1. Switch/button
2. Relay
3. Led
4. Power monitoring 3 pins (Sel,CF .CF1)
Step 3. Soldered those noted pads with suitable input/output pins on wemos d1.

lroubinov, 2020-09-27:
Thx for the replay
I have the same device as you having
can you save me some time and tell me exactly which pin on the tw-02 you connected to the wemos d1?

sigma-guilhfer, 2020-09-27:
[image]

I dont know wich one would be connected on the TW, I dont have them anymore, also I didnt use RX.
This is the best photo I took when I did this. For u to reset you need to press the button for 40sec.

lroubinov, 2020-09-27:
thx alot

lroubinov, 2020-09-27:
@sigma-guilhfer 
according the picture you connect something to gpio15
but in tasmota you didn't set anything to gpio15
am i missing something?

sigma-guilhfer, 2020-09-27:
> @sigma-guilhfer
> according the picture you connect something to gpio15
> but in tasmota you didn't set anything to gpio15
> am i missing something?

Sorry, I uploaded the worng image. Please, see my edited answer

lroubinov, 2020-09-27:
> > @sigma-guilhfer
> > according the picture you connect something to gpio15
> > but in tasmota you didn't set anything to gpio15
> > am i missing something?
> 
> Sorry, I uploaded the worng image. Please, see my edited answer

that's seems complicated... what all those resistors for?
according @ceaswaran picture he connect them all straight forward to the wemos
Is it because you use esp12f?

sigma-guilhfer, 2020-09-27:
I opened my device and this is how it is wired:

- gpio 16
- gpio 12        
- none            - none
- gpio 0          - gpio14
- gnd              - gpio 4
- vcc               - gpio 5

I dont think  that the resistors are necessary, its only to protect the gpio, but you need to follow the instructions from the vendor for it to work:

See page 8: https://docs.ai-thinker.com/_media/esp8266/docs/esp-12f_product_specification_en.pdf
It all depends on the module you are using, for the wemos I think they wired it already on the module.

lroubinov, 2020-09-28:
I soldered everything 
the relay, button and led working fine
but i can't get power monitoring reading... 
any idea?

ceaswaran, 2020-09-28:
For PM make sure u mapped the BL0937 pins correctly. (Ensure using  multimeter continuity mode )
Since BL0937 clone of  HLW8012 u can ref below pic , try tapping pin 6,7,8
Ref: https://user-images.githubusercontent.com/36055354/38105799-4a922bf2-338d-11e8-884c-d8e40ff15833.jpg

sigma-guilhfer, 2020-09-28:
Also make sure how your circuit works, the letter "i" stands for inverted (high/low), I had to invert most of the components logically for it to work, try to re-invert it.

After that, you must calibrate the sensor into the console on the device webpage or through mqtt. To do so, grab a multimeter and look at Tasmota web site.

ceaswaran, 2020-09-28:
Yes in my experience of using PM plugs all are SELi Only. :)

lroubinov, 2020-09-28:
The wired thing is i can't mapped the BL0937 
I tried using multimeter i get beep sound until the resistor and not till the bl0937
that's on all pins

M-Ahmad-Naeem, 2020-10-13:
> Managed to remove the TW02 module & Soldered Wemos D1 Mini
> I was able to map the Pins & Get all parameters working . Even the Power Monitoring works great.
> [image]
> [image]

@ceaswaran 

I am trying to do the same thing, I am actually here, can you please tell how did you manage to read all the parameters?

M-Ahmad-Naeem, 2020-10-13:
> > For PM make sure u mapped the BL0937 pins correctly. (Ensure using multimeter continuity mode )
> > Since BL0937 clone of HLW8012 u can ref below pic , try tapping pin 6,7,8
> > Ref: https://user-images.githubusercontent.com/36055354/38105799-4a922bf2-338d-11e8-884c-d8e40ff15833.jpg
> 
> @M-Ahmad-Naeem I hope you are looking for above explain, hope it helps

@ceaswaran I am asking, from picture I can see you have set the module to be generic "Generic Module". Please let me know what pins you have configured and how did you configured these pins.
Sorry for such a basic question!!

ceaswaran, 2020-10-13:
> @ceaswaran I am asking, from picture I can see you have set the module to be generic "Generic Module". Please let me know what pins you have configured and how did you configured these pins.
> Sorry [...]

Configuring pins on ESP8266/Wemos D1 mini is Individuals choice. u can configure based on whatever u feel ok
But Figuring out the TW-02 GPIO is thru multimeter continuity mode

Hope u got it. anyways pasting my Tasmota  config below:
```{"NAME":"TW02 TO ESP8266 PM","GPIO":[255,255,17,255,132,134,255,255,57,131,21,255,255],"FLAG":15,"BASE":18}```
[image]

M-Ahmad-Naeem, 2020-10-14:
> Configuring pins on ESP8266/Wemos D1 mini is Individuals choice. u can configure based on whatever u feel ok
> But Figuring out the TW-02 GPIO is thru multimeter continuity mode
>
> Hope u got it. [...]

Thank you so much, I will try and come back if I face any problem.

qntris, 2020-10-16:
Hi folks,

I am also trying to replace the TW-02 with a D1 mini, however I have two questions:
1. Although I can see the tracks (visible on the first picture below) going from the Sel,CF .CF1 pins of the BL0937 chip to the pins of the TW-02 board, when I probe them with continuity mode on my multimeter, I don't get any sound. Why would that be? The multimeter is not faulty. Judging by the tracks it seems that the connections are exactly as per @sigma-guilhfer 's [post](https://github.com/ct-Open-Source/tuya-convert/issues/572#issuecomment-699652793), with the little difference that on the right side (it is inverted on the picture so it is on the left) I don't have 4 but 5 pins (see the 3rd picture below).

2. Do you remove the TW-02 module, cut it or you directly solder the D1 mini to the existing pins on the wifi socket board and the TW-02? It seems that if the TW-02 is desoldered, there wouldn't be enough surface for soldering the cables from the D1 mini.

Below is the wifi switch that I have:

[image]
[image]
[image]

tarontop, 2020-10-17:
They seem to sell modules of esp8266

[https://pt.aliexpress.com/store/5790427](url)

tarontop, 2020-10-17:
They seem to sell modules of esp8266
[](https://pt.aliexpress.com/store/5790427)

Hendrikdenhond, 2020-10-20:
Hi Folks,

I solved the Tuya problem by removing the old chip and inserting an ESP 12-E chip.

My plug is an Elivco european plug.

I first flashed an esp12-E with the tasmota and configured the GPIO pin.
[image]

I soldered wires on the esp12-E chip for all needed GPIO pins, 3.3V and Ground.
Soldered a bridge between the enable pin and 3.3V
soldered a bridge between GPIO15 and GND.
[image]

Then i removed the wifi- board from the plug and identified the contacts, my plug was layed out as in the picture below.
[image]

After this is soldered the wires from the esp to the plug. 
[image]

the installed the circuitboard back into the plug.

[image]

I insulated the esp board with some electriciantape so that the board can not touch any of the exposed parts of the circuitboard.  I installed the esp board roughly in the same place as the old wifi board and closed the plug.  
[image]
I closed the plug with a little bit of hot glue.
and a working tasmota smart plug with power reading was born...

[image]

Hope this helps some of you guys.

blackest, 2020-10-28:
Hi I just opened up a jesled  led strip controller 4 wire from Amazon it runs on 24volts with a uk plug built in.  The cover has 5 recesses and squeezing from the cord end I was able to get them to release with no damage just using a fingernail.  The chip inside is marked TW-02 V2.0 and has a metal shield over the IC.  on 1 side i read 3v3 ,gnd,  rx , tx, pb16, pb15.   tx does not have a pad.  Unfortunately the main board is glued in with some kind of white resin liberally applied to 2 corners with board components partly covered. It looks impossible to remove without doing damage. So not recommended 

https://fccid.io/2ASQV-TW-02/User-Manual/User-manual-4211029.pdf

[image]

deiger, 2020-11-22:
@ceaswaran it seems it would be cleaner to replace with a [Tuya TYWE2S](https://developer.tuya.com/en/docs/iot/device-development/module/wifi-module/wifie2smodule?id=K9605u79tgxug), as it can be soldered right in place of the current module. The only place I could find it for sale though is [here](https://smartpets.en.made-in-china.com/product/ESzJbYnKVcRg/China-Esp-8285-Tuya-Tywe2s-Smart-WiFi-Module-Develop-Wireless-Networking-Intelligent-Module.html).

sigma-guilhfer, 2020-11-24:
> @ceaswaran it seems it would be cleaner to replace with a [Tuya TYWE2S](https://developer.tuya.com/en/docs/iot/device-development/module/wifi-module/wifie2smodule?id=K9605u79tgxug), as it can be [...]

Way easier! =)
Thank you!

CathedraCarlos, 2020-12-06:
Hi!

I'm trying to substitute a TW-02 in a plug with an esp8285 based ESP-M4. The contacts in my board are the same as @Hendrikdenhond  shows. I've configured all the inputs:
[image]

When I plug anything I get an accurate voltage reading, but no reading on power or current:
[image]

Any thoughts about what can be happening here?
I've double and triple checked the connections among BL0937 and my ESP

Hendrikdenhond, 2020-12-07:
Did you put a load (lamp, device, etc...) in your plug?

Did you connect the contacts of the plug to the right I/O on your board based on your configuration?
Config of your tasmota looks different from mine...

CathedraCarlos, 2020-12-07:
When the screen capture was done, a 50W led focus was connected to the plug. When the relay is open, the focus lights and the voltage reading begins, but nothing on power or current.

I've checked several times all the connections with a multimeter from the pins of the BL0937 to the GPIOs I've assigned. I've tested changing the GPIOs used in my ESP. At the very beginning I assigned the button to GPIO13, but I moved it to GPIO0 just to have another free GPIO and moved the SEL to GPIO13, playing with GPIOs 4, 5 and 12 with CF and CF1.

I think the configuration screen was modified in 9.1.0 tasmota version. Just in case I'm gonna try flashing an elder version (8.4.0)

Hendrikdenhond, 2020-12-07:
As the voltage and current reading uses the same pin (CF1)but it changed by the selection (sel) pin. I would take a close look if the SEL pin and the CF pin are connected good and set to the right gpio.

CathedraCarlos, 2020-12-07:
I think I've just found a bug in the 9.1.0 version. I've flashed the 8.4.0 version and now it works flawless
[image]

potados99, 2021-04-28:
Hi!

I've recently got a 12V LED dimmer from [aliexpress](https://www.aliexpress.com/item/1005001670745785.html?spm=a2g0o.productlist.0.0.64e97168kFy1vw&algo_pvid=e9b71f1d-73bf-424a-873d-0c4b18f0dc3b&algo_expid=e9b71f1d-73bf-424a-873d-0c4b18f0dc3b-0&btsid=0b0a556616191172241503727e066b&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_). I opened it up, and fount that it had [TW-03](https://www.electrodragon.com/product/w600-wifi-module-tw03-cortex-m3-1mb/) inside. I felt no hope in re-flashing the MCU for use of tuya-convert and ESP Home, so I decided to replace it with my ESP-01.

[image]

[image]

The LED was driven by an N-channel MOSFET([AO3414](http://aosmd.com/pdfs/datasheet/AO3414.pdf)), and the gate of the FET was controlled by `PB8` pin of the MCU. The pin was also pulled down to ground. Since the GPIO of the ESP will be `low` at the boot due to the pull-down, I used GPIO3(Rx) to control the FET. 

[image]

[image]

[image]

[image]

I desoldered the TW-03 and placed ESP-01 with proper connections. And it worked.

Hope this help you guys, and thank you @Hendrikdenhond for encouraging me!

satbeppo, 2022-01-19:
> Way easier! =) Thank you!

Hi, I replaced with TYWE2S, but it don't work. Anyone who got a Template for it? 
Thanks a lot!

jtauscher, 2023-03-01:
If anyone is interested in reflashing the TW-02 / W600 Chip with "Tasmota-like" firmware, I did it like this:

1. Download wm_tools from [https://github.com/w600/wm_tools](url)
2. Get firmware e.g. "OpenW600_1.15.514.fls" from [https://github.com/openshwprojects/OpenBK7231T_App/releases](url) for w600
3. Connect 3.3V (or 5V before voltage regulator) , GND, Rx/Tx to any UART converter
4. Open CMD and erase chip like `C:\tmp>wm_tools.exe -p COM4 erase_flash` . (For me it only worked _without_ specifying baud rate in a explicit way)
5. Flash firmware like `C:\tmp>wm_tools.exe -p COM4 write_flash OpenW600_1.15.514.fls`. If it returns error, either reset device or try erasing chip again.

Pin config in W600:

PA5 -> Btn
PB8 -> LED_n
PB13 -> BL0937CF1
PB15 -> Relay
PB16 -> BL0937SEL
PB18 -> BL0937CF

For CB2S Chips check the openBK project: [https://github.com/openshwprojects/OpenBK7231T_App](url)

TheOneOgre, 2024-12-07:
> If anyone is interested in reflashing the TW-02 / W600 Chip with "Tasmota-like" firmware, I did it like this:
>
> 1. Download wm_tools from [https://github.com/w600/wm_tools](url)
> 2. Get firmware e.g. [...]

Just wanted to confirm to future people that find this thread. This does indeed work just fine. Performed it on some recent Tuya plugs I got that the relay died in, so opened it up anyways to tinker, got the webpage working and connected to wifi just fine!

andy9a9, 2025-02-02:
Hi @jtauscher 
Where did you find this configuration:
> Pin config in W600:
>
> PA5 -> Btn
> PB8 -> LED_n
> PB13 -> BL0937CF1
> PB15 -> Relay
> PB16 -> BL0937SEL
> PB18 -> BL0937CF

especially the `PA5` pin, which is not mentioned in DS. After reverse-engineering, my `BTN` is connected to `RXD0`. But on the Tasmota config page is no `RXD0`. So how to handle it?

[3 more replies]