ct-Open-Source/tuya-convert #573: Bricked? Neo Smart Plug

QbaF, 2020-03-25:
Hi,

I tried to flash my Smart Plug Neo 3680W, 16A with power monitoring (working previously on Tuya cloud) with WIFi and MCU version 1.0.3
Made 3-4 flashing attempts. Everyone ends up with error `Device did not appear with the intermediate firmware`
https://pastebin.com/ecJVHGMM. 

After last one device stopped responding. No reaction on button, no LED light, no power delivery.

here are the logs
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4381913/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4381915/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4381916/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4381917/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4381918/smarthack-web.log)

Is it bricked and is it able to be restored?

kueblc, 2020-03-26:
It sounds like your device is stuck on the intermediate firmware. I can see from your web log that it did download the intermediate.

Try running the script again, but when you get to the point where it asks you to connect a phone and put the device in pairing mode, don't, just plug the device in and press enter.

QbaF, 2020-03-26:
Thanks. Tried 3 times. No success.

QbaF, 2020-03-27:
@kueblc do You think it could be because of 
a) operating this plug previously in Tuya Smart Life? (I didn't notice any upgrade)
b) using tuya-convert on Raspbian with installed Home Assistant?
c) connecting mobile to "vtrust-flash" was unstable? (a lot of attempts to lo in before connected)

I am asking because I do have more plugs like that and I am not sure how big are the chances to brick next ones.

kueblc, 2020-03-27:
> But what interesting - when I plug on the smart plug, I see WiFi "vtrust-recovery" instead of "vtrust-flash". Sometimes both. Could it be helpful?

This confirms that the device is still running the intermediate firmware. To recover, you will need to power it up when `vtrust-flash` is accessible.

If you are having unstable connection issues, you might want to try another platform, such as a laptop with Ubuntu, or another (clean) OS on your RPi.

QbaF, 2020-03-28:
I did try 4 android devices and macOS laptop and also (as mentioned with new logs) full clean RPi installation. 
Do You think WemosD1mini or NodeMCU can help?

QbaF, 2020-03-28:
OK, problem solved.

For others - that was a problem with WiFi on RPI4 caused by insufficient power as I use Conbee II and SSD. Previously I did not use WiFi so I did not realize there is something wrong and was not helpful that "vtrust-flash" appears. 

@kueblc thank for all the support.

closed 2020-03-28 by QbaF