ct-Open-Source/tuya-convert #574: could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC

mshuflin, 2020-03-26:
I am trying to flash a BSD48 smart socket (https://templates.blakadder.com/BSD48.html) and I receive this error:
Device did not appear with the intermediate firmware

Looking in the log, the item that looks relevant to me is in the psk log:

> new client on port 443 from 10.42.42.26:825
> ID: 0242416f68626d643661473931494652314fd6cc2e3a668ad2af0144f6e67ea04d8d972618c9f06e1727a58503babf424f
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4387397/smarthack-udp.log)

> PSK: 244fa34d5623e2b1c598a2b44ecfae632440154955249303a6c6c0dc1fcc00d4
> could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
> new client on port 443 from 10.42.42.26:1396

This is my first attempt at tuya-convert, thanks for all the work done to make this possible!

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4387406/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4387407/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4387411/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4387413/smarthack-web.log)

[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4387405/smarthack-wifi.log)

kueblc, 2020-03-26:
Unfortunately it looks like you have a patched firmware that is not compatible.

Follow #483 for more information and hopefully an eventual resolution.

closed 2020-03-26 by kueblc

AKSKMY, 2023-10-29:
Hi, @kueblc. I happen to have this plug as well… trying to do a smart home project with it and a ESP32. Tiny Circuit probably acting as the MQTT sender. 

However yeah same issue “firmware too new”. Is there a workaround?